lowRISC/ibex #1206: Single Step does not work, if BRANCHPREDICTOR=1.

Sarafim, 2020-12-01:
Hi, 
could you please check whether single step works, if BRANCHPREDICTOR=1.
It seems like depc_en is set to early, and that is why depc_q doesn't change its value.

I am asking, because on my side it works only if  BRANCHPREDICTOR=0.

GregAC, 2020-12-01:
It's certainly possible it's broken currently. The config with branch prediction enabled has seen the least testing out of all of the configs.

Do you observe single stepping failing always or only in certain scenarios?

Sarafim, 2020-12-01:
I can't say that I have tried lot of scenarios, but I see it always

GregAC, 2020-12-01:
When you say 'does not work' could you describe in more detail what you're actually seeing? Preferably with a waveform included.

Sarafim, 2020-12-01:
I mean that the depc_q does not change its value after the step
Correct behavior:
[image]
Wrong behavior:
[image]

vogelpi, 2021-04-13:
Hi @Sarafim ,
it has been a while but thanks for sharing the waveforms. Looking at them and the RTL code it's still not obvious what exactly causes the problem. When single stepping, the value saved into `depc` is always the PC in IF - independent whether the branch predictor is there or not. What changes with the branch predictor is that the PC in IF can either come from the skid buffer or the prefetch buffer.

To understand whether the problem is in the skid buffer or the prefetch buffer, some more insight into the test case would be needed. For example: What instructions are actually executed? Are you single stepping over branch instructions?

It would be very helpful if you could share a VCD file together with the disassembly. As @GregAC noted it is very well possible that there are issues around the branch predictor as its not as well tested as other configurations. But we would be interested to solve the issue.

rswarbrick, 2021-05-21:
Hi there! I'm closing this issue because it has been silent for a while and the development team doesn't have enough information to make any progress. @Sarafim: if you manage to reproduce this and can share a trace and disassembly as @vogelpi suggested, please feel free to re-open.

closed 2021-05-21 by rswarbrick