lowRISC/ibex #1209: Ibex Simple System build error

kidonglee, 2020-12-02:
Hi, 

I got error messages while building 'Simple System' as below.
Please help.

Thanks in advance.

`
root@0184358265d9:~/ibex# fusesoc --cores-root=. run --target=sim --setup --build lowrisc:ibex:ibex_simple_system --RV32E=0 --RV32M=ibex_pkg::RV32MFast
WARNING: Unknown item formal in section Target
WARNING: Unknown item formal in section Target
WARNING: Unknown item formal in section Target
INFO: Preparing lowrisc:dv_verilator:ibex_pcounts:0
INFO: Preparing lowrisc:dv_verilator:memutil_dpi:0
INFO: Preparing lowrisc:dv_verilator:simutil_verilator:0
INFO: Preparing lowrisc:ibex:ibex_pkg:0.1
INFO: Preparing lowrisc:prim:assert:0.1
INFO: Preparing lowrisc:prim:primgen:0.1
INFO: Preparing lowrisc:prim:secded:0.1
INFO: Preparing lowrisc:tool:check_tool_requirements:0.1
INFO: Preparing lowrisc:dv_verilator:memutil_verilator:0
INFO: Preparing lowrisc:ibex:ibex_tracer:0.1
INFO: Preparing lowrisc:lint:common:0.1
INFO: Preparing lowrisc:prim:prim_pkg:0.1
INFO: Preparing lowrisc:prim:clock_gating:0
INFO: Preparing lowrisc:prim:lfsr:0.1
INFO: Preparing lowrisc:prim:ram_1p:0
INFO: Preparing lowrisc:prim:ram_2p:0
INFO: Preparing lowrisc:ibex:ibex_icache:0.1
INFO: Preparing lowrisc:ibex:sim_shared:0
INFO: Preparing lowrisc:ibex:ibex_core:0.1
INFO: Preparing lowrisc:ibex:ibex_core_tracing:0.1
INFO: Preparing lowrisc:ibex:ibex_simple_system:0
INFO: Generating lowrisc:prim:prim_pkg-impl:0.1
Creating prim_pkg.sv
Core file written to prim_pkg.core.
INFO: Generating lowrisc:prim:clock_gating-impl:0
Implementations for primitive clock_gating: xilinx, generic
Inspecting generic module /root/ibex/vendor/lowrisc_ip/ip/prim_generic/rtl/prim_generic_clock_gating.sv
Creating SystemVerilog module for abstract primitive
Abstract primitive written to /root/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_clock_gating-impl_0/prim_clock_gating.sv
Creating core file for primitive clock_gating.
Core file written to /root/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_clock_gating-impl_0/prim_clock_gating.core
INFO: Generating lowrisc:prim:ram_1p-impl:0
Implementations for primitive ram_1p: badbit, generic
Inspecting generic module /root/ibex/vendor/lowrisc_ip/ip/prim_generic/rtl/prim_generic_ram_1p.sv
Creating SystemVerilog module for abstract primitive
Abstract primitive written to /root/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_ram_1p-impl_0/prim_ram_1p.sv
Creating core file for primitive ram_1p.
Core file written to /root/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_ram_1p-impl_0/prim_ram_1p.core
INFO: Generating lowrisc:prim:ram_2p-impl:0
Implementations for primitive ram_2p: generic
Inspecting generic module /root/ibex/vendor/lowrisc_ip/ip/prim_generic/rtl/prim_generic_ram_2p.sv
Creating SystemVerilog module for abstract primitive
Abstract primitive written to /root/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_ram_2p-impl_0/prim_ram_2p.sv
Creating core file for primitive ram_2p.
Core file written to /root/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_ram_2p-impl_0/prim_ram_2p.core
INFO: Wrote dependency graph to /root/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/lowrisc_ibex_ibex_simple_system_0.deps-after-generators.dot
INFO: Wrote Makefile fragment to /root/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/core-deps.mk
INFO: Setting up project

INFO: Running pre_build script check_tool_requirements
INFO: Found sufficiently recent version of verilator (found 4.104, need 4.028)
INFO: Found sufficiently recent version of edalize (found 0.2.2, need 0.2.0)
INFO: Building simulation model
verilator -f lowrisc_ibex_ibex_simple_system_0.vc --trace --trace-fst --trace-structs --trace-params --trace-max-array 1024 -CFLAGS "-std=c++11 -Wall -DVM_TRACE_FMT_FST -DTOPLEVEL_NAME=ibex_simple_system -g" -LDFLAGS "-pthread -lutil -lelf" -Wall --unroll-count 72
make  -f Vibex_simple_system.mk
make[1]: Entering directory '/root/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator'
ccache g++  -I. -Wall -Wextra -Wfloat-conversion -Wlogical-op -Werror -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -I../src/lowrisc_dv_verilator_ibex_pcounts_0/cpp -I../src/lowrisc_dv_verilator_memutil_dpi_0/cpp -I../src/lowrisc_dv_verilator_simutil_verilator_0/cpp -I../src/lowrisc_prim_assert_0.1/rtl -I../src/lowrisc_prim_util_memload_0/rtl -I../src/lowrisc_dv_verilator_memutil_verilator_0/cpp -std=c++11 -Wall -DVM_TRACE_FMT_FST -DTOPLEVEL_NAME=ibex_simple_system -g  -std=gnu++14 -Os -c -o ibex_pcounts.o ../src/lowrisc_dv_verilator_ibex_pcounts_0/cpp/ibex_pcounts.cc
ccache g++  -I. -Wall -Wextra -Wfloat-conversion -Wlogical-op -Werror -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -I../src/lowrisc_dv_verilator_ibex_pcounts_0/cpp -I../src/lowrisc_dv_verilator_memutil_dpi_0/cpp -I../src/lowrisc_dv_verilator_simutil_verilator_0/cpp -I../src/lowrisc_prim_assert_0.1/rtl -I../src/lowrisc_prim_util_memload_0/rtl -I../src/lowrisc_dv_verilator_memutil_verilator_0/cpp -std=c++11 -Wall -DVM_TRACE_FMT_FST -DTOPLEVEL_NAME=ibex_simple_system -g  -std=gnu++14 -Os -c -o dpi_memutil.o ../src/lowrisc_dv_verilator_memutil_dpi_0/cpp/dpi_memutil.cc
[... 67 lines cut ...]
             = __Vdlyvval__ibex_simple_system__DOT__u_core__DOT__u_ibex_core__DOT__rvfi_stage_intr__v0;
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Vibex_simple_system.cpp:2582:13: error: '__Vdlyvval__ibex_simple_system__DOT__u_core__DOT__u_ibex_core__DOT__rvfi_stage_rs1_addr__v0' may be used uninitialized in this function [-Werror=maybe-uninitialized]
         vlTOPp->ibex_simple_system__DOT__u_core__DOT__u_ibex_core__DOT__rvfi_stage_rs1_addr[0U]
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
             = __Vdlyvval__ibex_simple_system__DOT__u_core__DOT__u_ibex_core__DOT__rvfi_stage_rs1_addr__v0;
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Vibex_simple_system.cpp:2589:13: error: '__Vdlyvval__ibex_simple_system__DOT__u_core__DOT__u_ibex_core__DOT__rvfi_stage_rs2_addr__v0' may be used uninitialized in this function [-Werror=maybe-uninitialized]
         vlTOPp->ibex_simple_system__DOT__u_core__DOT__u_ibex_core__DOT__rvfi_stage_rs2_addr[0U]
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
             = __Vdlyvval__ibex_simple_system__DOT__u_core__DOT__u_ibex_core__DOT__rvfi_stage_rs2_addr__v0;
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Vibex_simple_system.cpp:2596:13: error: '__Vdlyvval__ibex_simple_system__DOT__u_core__DOT__u_ibex_core__DOT__rvfi_stage_trap__v0' may be used uninitialized in this function [-Werror=maybe-uninitialized]
         vlTOPp->ibex_simple_system__DOT__u_core__DOT__u_ibex_core__DOT__rvfi_stage_trap[0U]
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
             = __Vdlyvval__ibex_simple_system__DOT__u_core__DOT__u_ibex_core__DOT__rvfi_stage_trap__v0;
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
/usr/local/share/verilator/include/verilated.mk:235: recipe for target 'Vibex_simple_system.o' failed
make[1]: *** [Vibex_simple_system.o] Error 1
make[1]: Leaving directory '/root/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator'
Makefile:13: recipe for target 'Vibex_simple_system' failed
make: *** [Vibex_simple_system] Error 2
ERROR: Failed to build lowrisc:ibex:ibex_simple_system:0 : 'make' exited with an error code
`

rswarbrick, 2020-12-02:
I think you're probably using an oldish GCC. The "maybe uninitialised" warning used to trigger really easily and was often tickled by code generated by Verilator. As a quick hack to get things working, I think you can pass `-Wno-error=maybe-uninitialized` to GCC to avoid triggering an error. You can either do this by adding a command line flag to FuseSoc (but I can't remember it off-hand) or adding the flag to the list at the end of `example/simple_system/ibex_simple_system.core`.

If this fixes things for you, can you tell us your GCC version? I think we either need to change our flags to allow this, or update the documentation to require a newer GCC.

kidonglee, 2020-12-02:
Dear rswarbrick, 

Thank you for the advice.
I added the flag '-Wno-error=maybe-uninitialized' to the file 'ibex_simple_system.core'.
However I still have the same error messages.
I attach the log below. Please check if the flag added is correct or not.

And the version of GCC is as follows,
```
root@0184358265d9:~/ibex# g++ --version
g++ (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

root@0184358265d9:~/ibex# gcc --version
gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```
**[log]**
```
...
ccache g++  -I. -Wall -Wextra -Wfloat-conversion -Wlogical-op -Werror -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -I../src/lowrisc_dv_verilator_ibex_pcounts_0/cpp -I../src/lowrisc_dv_verilator_memutil_dpi_0/cpp -I../src/lowrisc_dv_verilator_simutil_verilator_0/cpp -I../src/lowrisc_prim_assert_0.1/rtl -I../src/lowrisc_prim_util_memload_0/rtl -I../src/lowrisc_dv_verilator_memutil_verilator_0/cpp -std=c++11 -Wall -DVM_TRACE_FMT_FST -DTOPLEVEL_NAME=ibex_simple_system -g -Wno-error=maybe-uninitialized  -std=gnu++14  -c -o Vibex_simple_system__Syms.o Vibex_simple_system__Syms.cpp
ccache g++  -I. -Wall -Wextra -Wfloat-conversion -Wlogical-op -Werror -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -I../src/lowrisc_dv_verilator_ibex_pcounts_0/cpp -I../src/lowrisc_dv_verilator_memutil_dpi_0/cpp -I../src/lowrisc_dv_verilator_simutil_verilator_0/cpp -I../src/lowrisc_prim_assert_0.1/rtl -I../src/lowrisc_prim_util_memload_0/rtl -I../src/lowrisc_dv_verilator_memutil_verilator_0/cpp -std=c++11 -Wall -DVM_TRACE_FMT_FST -DTOPLEVEL_NAME=ibex_simple_system -g -Wno-error=maybe-uninitialized  -std=gnu++14  -c -o Vibex_simple_system__Trace__Slow.o Vibex_simple_system__Trace__Slow.cpp
Vibex_simple_system.cpp: In static member function 'static void Vibex_simple_system::_sequent__TOP__11(Vibex_simple_system__Syms*)':
Vibex_simple_system.cpp:2659:13: warning: '__Vdlyvval__ibex_simple_system__DOT__u_core__DOT__u_ibex_core__DOT__id_stage_i__DOT__imd_val_q__v2' may be used uninitialized in this function [-Wmaybe-uninitialized]
         vlTOPp->ibex_simple_system__DOT__u_core__DOT__u_ibex_core__DOT__id_stage_i__DOT__imd_val_q[1U]
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
             = __Vdlyvval__ibex_simple_system__DOT__u_core__DOT__u_ibex_core__DOT__id_stage_i__DOT__imd_val_q__v2;
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Vibex_simple_system.cpp:2561:13: warning: '__Vdlyvval__ibex_simple_system__DOT__u_core__DOT__u_ibex_core__DOT__rvfi_stage_order__v0' may be used uninitialized in this function [-Wmaybe-uninitialized]
         vlTOPp->ibex_simple_system__DOT__u_core__DOT__u_ibex_core__DOT__rvfi_stage_order[0U]
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
             = __Vdlyvval__ibex_simple_system__DOT__u_core__DOT__u_ibex_core__DOT__rvfi_stage_order__v0;
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Vibex_simple_system.cpp:2519:13: warning: '__Vdlyvval__ibex_simple_system__DOT__u_core__DOT__u_ibex_core__DOT__rvfi_stage_mem_addr__v0' may be used uninitialized in this function [-Wmaybe-uninitialized]
         vlTOPp->ibex_simple_system__DOT__u_core__DOT__u_ibex_core__DOT__rvfi_stage_mem_addr[0U]
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
             = __Vdlyvval__ibex_simple_system__DOT__u_core__DOT__u_ibex_core__DOT__rvfi_stage_mem_addr__v0;
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
...
```

rswarbrick, 2020-12-02:
Does the build fail? These are warnings emitted by the compiler, which `-Werror` turned into errors, causing the build to fail. Passing `-Wno-error=maybe-uninitialized` turns off this promotion so that the build should succeed.

Thanks for giving your GCC version. Looks like 7.5 was only released last November (wow, they're getting through major versions quickly!), so maybe we want to support it. @imphil, what do you think?

kidonglee, 2020-12-02:
Dear rswarbrick, 

I solved my problem.
I add the flags '-Wno-maybe-uninitialized' instead of ''-Wno-error=maybe-uninitialized.

Thanks for your help.

closed 2020-12-02 by kidonglee