lowRISC/ibex #1211: Purpose of Request Outstanding Queue?

sajjadahmed677, 2020-12-07:
I am looking into prefetch_buffer for understanding the fetch operation of ibex. Here i see a "request outstanding queue" but didn't get the clear image of it. why we need such queue? what if we don't use that?

rswarbrick, 2020-12-08:
This is a queue of requests that have been made on the (pipelined) instruction bus. If we make lots of requests, we need to know to wait for the answers (so that we know what responses from the bus mean). Also, we have to queue up the results so that we can handle the case where we've fetched ahead of the processor (the whole point of the prefetch buffer) and the responses come back before the processor can use them.

imphil, 2021-01-07:
@sajjadahmed677  I think your questions should be answered. If you have follow-up questions, please re-open this issue. Otherwise open a new one. Thanks for your interest in Ibex!

closed 2021-01-07 by imphil