lowRISC/ibex #1213: Syntax error in assert line 836 of file ibex_controller.sv

Fresher14, 2020-12-15:
Current code:
`ASSERT(AlwaysInstrClearOnMispredict, nt_branch_mispredict_o -> instr_valid_clear_o)

Suggested code (a 'pipe'='|' added):
`ASSERT(AlwaysInstrClearOnMispredict, nt_branch_mispredict_o |-> instr_valid_clear_o)

Current code fails with Xcellium 19.03-s001 for example. But works with a more recent version (20.10...)

tomroberts-lowrisc, 2020-12-15:
Thanks for the report @Fresher14 - can you create a PR to fix this?

rswarbrick, 2020-12-15:
There's a difference between the two expressions (although the two are equivalent here). The code at the moment uses `->`, which is a binary logical operator, and can be used to construct SV expressions. You're suggesting replacing it with `|->`, the overlapped implication operator. This can only be used in SystemVerilog assertions.

Are you saying that Xcelium 19.03 doesn't support the `->` operator? (I'm not sure when it was added to the SystemVerilog specification, but it appears in IEEE 1800-2017, so presumably before 2017!)

Fresher14, 2020-12-15:
I confirm that this code fails with Xcellium 19.03-s001 but not with 20.10_i696.
I take your word on assertions. I'm too light on this topic.

rswarbrick, 2020-12-15:
Oh, sorry, I wasn't trying to suggest you rewrote it as `B|!A`. Do we claim to support this version of Xcelium? (@imphil / @tomroberts-lowrisc ?). If so, we should switch to overlapping implication, as you suggest.

tomroberts-lowrisc, 2020-12-15:
I think we should just switch to the overlapping implication anyway as that's probably what was intended

Fresher14, 2020-12-16:
Overlapping implication is the right choice then. I will change on my side.
No clue on how to push this modification here in GitHub. I assume one of you will do it.

rswarbrick, 2020-12-16:
We use the usual GitHub flow with pull requests. There's reasonably comprehensive documentation written by GitHub here: https://docs.github.com/en/free-pro-team@latest/github/collaborating-with-issues-and-pull-requests. You'll also find (rather terse) some notes about commit messages in [`CONTRIBUTING.md`](https://github.com/lowRISC/ibex/blob/master/CONTRIBUTING.md) at the top of the repository.

As Tom said earlier, it would be great if you fancied getting involved by creating a pull request to fix this.

Fresher14, 2020-12-16:
It looks like I do not have the access rights:

ERROR: Permission to lowRISC/ibex.git denied to Fresher14.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights and the repository exists.

rswarbrick, 2020-12-16:
Yep, you need to fork the repository and create a pull request, as described in the document I linked to above.

Fresher14, 2020-12-16:
'Fork' was lacking indeed. Now modification is pending approval: no more in my hands.

commit 0199bbae (2020-12-17): Use overlapping implications for Xcelium 19.03

Xcelium 19.03 doesn't support the `->` (binary logical) operator, but
using `|->` (overlapping implication) is equivalent here and supported;
use this operator instead.

Fixes #1213
closed 2020-12-17 by imphil