lowRISC/ibex #1214: Ask a question related to Ibex

diakovd, 2020-12-15:
## My Environment

**EDA tool and version:**

Quartus, modelsim, Segger Emdeded studio

**Operating system:**

**Version of the Ibex source code:**

Hi , 
I have some question and finde some bug when using Ibex:

1.What for   0x80 in code: PC_BOOT: fetch_addr_n = { boot_addr_i[31:8], 8'h80 }; ?
I set 0x80 to 0x00 in boot address  as this cause error at initialization  copying constan variable from ROM to RAM with Segger Embeded studio.
  
2.For compiling in Quartus Prime I was replase all generate block to this coding style  
  generate
  genvar k;
  // bit reverse operand_a for left shifts and bit counting
  for (k = 0; k < 32; k++) begin : gen_rev_operand_a
    assign operand_a_rev[k] = operand_a_i[31-k];
  end
  endgenerate

3. Fix bug After interrupt innstruction jump to wrong exeption address:
   in file ibex_cs_registers.sv
   CSR_MTVEC: mtvec_d =  {csr_wdata_int[31:8], 6'b0, 2'b01}; replace to CSR_MTVEC: mtvec_d =  csr_wdata_int;
   in file ibex_if_stage.sv
   EXC_PC_IRQ:     exc_pc = { csr_mtvec_i[31:8], 1'b0, irq_id[4:0], 2'b00 }; replace to EXC_PC_IRQ:     exc_pc = csr_mtvec_i;

Thank you,
Dmitry

tomroberts-lowrisc, 2020-12-16:
I will try to answer these questions below:
1. We define the reset vector to be the trap base + 0x80 (effectively the top of the vector table). I suppose `boot_addr_i` is a slightly misleading name since it is specifying the trap base address.
2. This was discussed in [this](https://github.com/lowRISC/ibex/issues/117) issue. The conclusion: we won't change the rtl to work around old / unsupported EDA tools.
3. Ibex always uses vectored interrupt handling (see the `mtvec` section of the RISCV privileged spec). This is why bit[0] is is set, and the irq id is used to form the trap address.

imphil, 2021-01-07:
@diakovd I think your questions should be answered. If you have follow-up questions, please re-open this issue. Otherwise open a new one. Thanks for your interest in Ibex!

closed 2021-01-07 by imphil