lowRISC/ibex #1217: riscv_machine_mode_rand_test fails with spike giving a large instruction count

sammy17, 2020-12-17:
## My Environment

**EDA tool and version:**
VCS Q-2020

**Operating system:**
Linux 3.10.0-1127.19.1.el7.x86_64

**Version of the Ibex source code:**
Using the master branch HEAD=aa7582b8d420888541c2e8688447c17918a71aeb

## **Issue observed:**
I am trying to run the verification flow of ibex core with riscv-dv by following the instructions at https://github.com/lowRISC/ibex/blob/master/doc/03_reference/verification.rst. 
After setting up the tools I tried running:

`make TEST=riscv_machine_mode_rand_test ITERATIONS=1`

I first tried with the lowrisc pre-compiled gnu-toolchain and lowrisc specific branch of spike as instructed, but got the following error:
```
spike --log-commits --isa=rv32imcb -l out/seed-4388/instr_gen/asm_tests/riscv_machine_mode_rand_test_0.o                                                                        
error: bad --isa option 'rv32imcb'. unsupported extension 'b'
```

Then I tried with the locally built RISC-V GNU toolchain and Spike (got them from respective github repos). I went past above error with these toolchains but still **riscv_machine_mode_rand_test**  failed with a mismatch. Issue in this case is Spike executes for too long and gets terminated due to timeout. I notice that the instruction count from Spike is really high compared to the RTL simulation. Please find the terminal output below:

```
Thu, 17 Dec 2020 04:29:42 ERROR    Timeout[10s]: /projectnb/risc-v/chath/riscv-gnu-toolchain/build/riscv/bin/spike --log-commits --isa=rv32imcb -l out/seed-29116/instr_gen/asm_tests/riscv_machine_mode_rand_test_0.o &> out/seed-29116/instr_gen/spike_sim/riscv_machine_mode_rand_test.0.log
rm -rf out/seed-29116/rtl_sim
cp -r out/rtl_sim out/seed-29116
Thu, 17 Dec 2020 04:29:44 INFO     Processing simulator setup file : yaml/rtl_simulation.yaml
Thu, 17 Dec 2020 04:29:44 INFO     Processing regression test list : riscv_dv_extension/testlist.yaml, test: riscv_machine_mode_rand_test
Thu, 17 Dec 2020 04:29:44 INFO     Found matched tests: riscv_machine_mode_rand_test, iterations:1
Thu, 17 Dec 2020 04:29:44 INFO     Running RTL simulation...
Thu, 17 Dec 2020 04:29:44 INFO     Running core_ibex_base_test with out/seed-29116/instr_gen/asm_tests/riscv_machine_mode_rand_test_0.bin
Thu, 17 Dec 2020 04:29:55 INFO     Processing regression test list : riscv_dv_extension/testlist.yaml, test: riscv_machine_mode_rand_test
Thu, 17 Dec 2020 04:29:55 INFO     Found matched tests: riscv_machine_mode_rand_test, iterations:1
Thu, 17 Dec 2020 04:29:55 INFO     Comparing spike/DUT sim result : out/seed-29116/instr_gen/asm_tests/riscv_machine_mode_rand_test.0.o
Thu, 17 Dec 2020 04:29:55 INFO     Processing ibex log : out/seed-29116/rtl_sim/riscv_machine_mode_rand_test.0/trace_core_00000000.log
Thu, 17 Dec 2020 04:29:55 INFO     Processed instruction count : 5279
Thu, 17 Dec 2020 04:29:55 INFO     CSV saved to : out/seed-29116/rtl_sim/riscv_machine_mode_rand_test.0/trace_core_00000000.csv
Thu, 17 Dec 2020 04:29:55 INFO     Processing spike log : out/seed-29116/instr_gen/spike_sim/riscv_machine_mode_rand_test.0.log
Thu, 17 Dec 2020 04:30:00 INFO     Processed instruction count : 1207102
Thu, 17 Dec 2020 04:30:00 INFO     CSV saved to : out/seed-29116/instr_gen/spike_sim/riscv_machine_mode_rand_test.0.csv
Thu, 17 Dec 2020 04:30:00 INFO     0 PASSED, 1 FAILED
Thu, 17 Dec 2020 04:30:00 INFO     RTL & ISS regression report at out/seed-29116/regr.log
```
**Please help to identify and resolve above errors.**

imphil, 2020-12-17:
It looks like your version of Spike isn't correct somehow. Can you show us a log what you did to compile spike? Which git commit hash of the Spike repository are you using?

sammy17, 2020-12-17:
Spike (lowRISC branch) git commit hash:
43003ea3ed9d2c1a1c060416da91ccf5b044c55c

Log file containing configure and build commands/outputs:
[log.txt](https://github.com/lowRISC/ibex/files/5710111/log.txt)

Let me know if you need any other info.

imphil, 2020-12-17:
You're building from the `master` branch of riscv-isa-sim, you need to build from the `ibex` branch (commit 255bf1cacc599b1413438c269100f3ecd0eb3352). Do a `git checkout ibex` in the riscv-isa-sim directory, and rebuild spike.

sammy17, 2020-12-17:
My bad, it worked after compiling with the ibex branch correclty. Thanks for the help!

closed 2020-12-17 by sammy17