lowRISC/ibex #1219: build simple system error

guqian110, 2021-01-07:
## My Environment

Hi all,

I have compiled verilator from the source and installed fusesoc. when I was building the simple system under the root of ibex directory,

```
make build-simple-system
```

it gives warning information as following,

```
fusesoc --cores-root=. run --target=sim --setup --build \
	lowrisc:ibex:ibex_simple_system \
	--RV32E=0 --RV32M=ibex_pkg::RV32MFast --RV32B=ibex_pkg::RV32BNone --RegFile=ibex_pkg::RegFileFF --BranchTargetALU=0 --WritebackStage=0 --ICache=0 --ICacheECC=0 --BranchPredictor=0 --PMPEnable=0 --PMPGranularity=0 --PMPNumRegions=4
WARNING: Unknown item formal in section Target
WARNING: Unknown item formal in section Target
WARNING: Unknown item formal in section Target
INFO: Preparing lowrisc:dv_verilator:ibex_pcounts:0
INFO: Preparing lowrisc:dv_verilator:memutil_dpi:0
INFO: Preparing lowrisc:dv_verilator:simutil_verilator:0
INFO: Preparing lowrisc:ibex:ibex_pkg:0.1
INFO: Preparing lowrisc:prim:assert:0.1
INFO: Preparing lowrisc:prim:primgen:0.1
INFO: Preparing lowrisc:prim:secded:0.1
INFO: Preparing lowrisc:tool:check_tool_requirements:0.1
INFO: Preparing lowrisc:dv_verilator:memutil_verilator:0
INFO: Preparing lowrisc:ibex:ibex_tracer:0.1
INFO: Preparing lowrisc:lint:common:0.1
INFO: Preparing lowrisc:prim:prim_pkg:0.1
INFO: Preparing lowrisc:prim:clock_gating:0
INFO: Preparing lowrisc:prim:lfsr:0.1
INFO: Preparing lowrisc:prim:ram_1p:0
INFO: Preparing lowrisc:prim:ram_2p:0
INFO: Preparing lowrisc:ibex:ibex_icache:0.1
INFO: Preparing lowrisc:ibex:sim_shared:0
INFO: Preparing lowrisc:ibex:ibex_core:0.1
INFO: Preparing lowrisc:ibex:ibex_core_tracing:0.1
INFO: Preparing lowrisc:ibex:ibex_simple_system:0
INFO: Generating lowrisc:prim:prim_pkg-impl:0.1
Creating prim_pkg.sv
Core file written to prim_pkg.core.
INFO: Generating lowrisc:prim:clock_gating-impl:0
Implementations for primitive clock_gating: generic, xilinx
Inspecting generic module /home/qian/repositories/ibex/vendor/lowrisc_ip/ip/prim_generic/rtl/prim_generic_clock_gating.sv
Creating SystemVerilog module for abstract primitive
Abstract primitive written to /home/qian/repositories/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_clock_gating-impl_0/prim_clock_gating.sv
Creating core file for primitive clock_gating.
Core file written to /home/qian/repositories/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_clock_gating-impl_0/prim_clock_gating.core
INFO: Generating lowrisc:prim:ram_1p-impl:0
Implementations for primitive ram_1p: generic, badbit
Inspecting generic module /home/qian/repositories/ibex/vendor/lowrisc_ip/ip/prim_generic/rtl/prim_generic_ram_1p.sv
Creating SystemVerilog module for abstract primitive
Abstract primitive written to /home/qian/repositories/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_ram_1p-impl_0/prim_ram_1p.sv
Creating core file for primitive ram_1p.
Core file written to /home/qian/repositories/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_ram_1p-impl_0/prim_ram_1p.core
INFO: Generating lowrisc:prim:ram_2p-impl:0
Implementations for primitive ram_2p: generic
Inspecting generic module /home/qian/repositories/ibex/vendor/lowrisc_ip/ip/prim_generic/rtl/prim_generic_ram_2p.sv
Creating SystemVerilog module for abstract primitive
Abstract primitive written to /home/qian/repositories/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_ram_2p-impl_0/prim_ram_2p.sv
Creating core file for primitive ram_2p.
Core file written to /home/qian/repositories/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/generated/lowrisc_prim_ram_2p-impl_0/prim_ram_2p.core
INFO: Wrote dependency graph to /home/qian/repositories/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/lowrisc_ibex_ibex_simple_system_0.deps-after-generators.dot
INFO: Wrote Makefile fragment to /home/qian/repositories/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/core-deps.mk
INFO: Setting up project

INFO: Running pre_build script check_tool_requirements
INFO: Found sufficiently recent version of verilator (found 4.107, need 4.028)
[... 40 lines cut ...]
      |   ^~~~~~~~~~~
%Warning-LATCH: ../src/lowrisc_ibex_sim_shared_0/./rtl/bus.sv:70:3: Latch inferred for signal 'ibex_simple_system.u_bus.device_sel_req' (not all control paths of combinational always assign a value)
                                                                  : ... Suggest use of always_latch for intentional latches
   70 |   always_comb begin
      |   ^~~~~~~~~~~
%Error: Exiting due to 10 warning(s)
Makefile:16: recipe for target 'Vibex_simple_system.mk' failed
make[1]: *** [Vibex_simple_system.mk] Error 1
make[1]: Leaving directory '/home/qian/repositories/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-verilator'
```

It seems there are too many warnings to stall the build process, how could I fix it ? Please help me,  thanks in advance.

**EDA tool and version:**

verilator version: Verilator 4.107 devel rev v4.106-155-ga1170027
fusesoc version: 1.11.dev57+g1fb8c93

**Operating system:**

Ubuntu 18.04 LTS

**Version of the Ibex source code:**

master branch(0199bbae665b9b142144e6688279e2ecef7d83a0)

imphil, 2021-01-07:
It seems like these are warnings that Verilator 4.107 added. You can remove the `-Wall` option at https://github.com/lowRISC/ibex/blob/master/examples/simple_system/ibex_simple_system.core#L168 to get around it for now. 

We'd also appreciate a pull request to fix those warnings. For the parameters, it just only be a matter of removing them if they are truly unused.

guqian110, 2021-01-08:
> It seems like these are warnings that Verilator 4.107 added. You can remove the `-Wall` option at https://github.com/lowRISC/ibex/blob/master/examples/simple_system/ibex_simple_system.core#L168 to [...]

Thank you for helping, I solved it by replacing `-Wall` with `-Wno-lint` , I'll try to work on these warnings later.

towoe, 2021-01-09:
Just for reference, I think it is also possible to use the Verilator option `-Wno-fatal` to not exits with an error if warnings occur.
Together with `-Wall` the warnings are still reported but the build continues.

towoe, 2021-01-11:
Thanks for reporting this @guqian110. Fixed with #1229, I tested it now with Verilator 4.109 devel rev v4.108-6-g4f1867a4

closed 2021-01-11 by towoe

guqian110, 2021-01-12:
> Thanks for reporting this @guqian110. Fixed with #1229, I tested it now with Verilator 4.109 devel rev v4.108-6-g4f1867a4

Great job!