lowRISC/ibex #1220: List of tests for each config

imphil, 2021-01-07:
We have multiple Ibex configurations currently, listed at https://github.com/lowRISC/ibex/blob/master/ibex_configs.yaml, which enable a different set of functionality.

To run full DV regressions (in CI), I need a list of tests which are expected to pass for each configuration. For example, the small configuration doesn't have PMP and shouldn't run the PMP tests.

Currently, we run `make -C dv/uvm/core_ibex ISS=spike IBEX_CONFIG=${{ config }}` (with `${{ config }}`) being replaced by one of the supported configurations). Can we somehow expand the tooling to also pick the right set of tests? I can see two options (there might be more):

* Add the test list to ibex_configs.yml
* Modify the tests to skip themselves when the right RTL parameter isn't set

The second options seems more flexible to me and it will also help users who don't run with a config, but just set the parameters directly.

Ideally, we can come up with a way to "discover" the required set of tests in some of the tooling we have in the repository, and don't hard-code this list into the CI configuration. This will allow human users to run (hopefully) a single command to get the right set of tests for their config.

@udinator @GregAC

udinator, 2021-01-08:
I think I agree with the second option here, I don't think that adding the static test list to each config is the right way of doing this.
Potentially a simple implementation method would look like this:
```console
- Add anew  list field to each YAML entry, containing the (if any) RTL parameters that it needs to run properly
- In sim.py, skip a test from being processed if any of its requisite parameters aren't present (probably a quick for loop)
```
One question I do have - this (or a similar approach) would probably require that the `sim.py` script have access to the list of parameter values from `ibex_configs.yaml.
@GregAC is this possible to do with the configs flow that you set up?

GregAC, 2021-01-11:
`sim.py` could read `ibex_configs.yaml` directly easily enough. The other option would be to pass config details into `sim.py` using `ibex_config.py`. `ibex_config.py` is designed to read the config YAML and produce appropriate arguments for different tools so we'd add a new option for it to generate appropriate `sim.py` arguments.

I'd tend towards the first option, `ibex_config.py` could be imported into `sim.py` so we don't have two different implementations of config file reading (there's very little to it but there's a default file specified and an environment variable override, plus if we do make things more complex in the future we don't want to separately implement new features in two different places).

udinator, 2021-03-04:
Sorry for my delay here, have been tied up with other OT work.

@GregAC I agree with the first option as well, I'll draft a sample implementation soon and get a PR ready.

imphil, 2021-03-04:
Great, thanks a lot @udinator!

GregAC, 2021-12-10:
This functionality has been added

closed 2021-12-10 by GregAC