lowRISC/ibex #1221: Run full DV regression regularly

imphil, 2021-01-07:
We currently run a very limited smoke regression on every master commit/PR, but don't run the full DV regression suite. This suite takes significantly longer to run, so we need to run it in regular intervals, e.g. once a day or a week, depending on the runtime and availability of compute resources and licenses. (I had an initial look and we probably need to run it on the weekend, we're busy 24 hours otherwise.)

The current plan is to run the tests in our normal Azure Pipelines infrastructure, scheduled according to our desired interval.

Dependencies:
* [ ] Choose a suitable test list for each configuration: #1220 
* [ ] Pipelines configuration: Current draft at https://github.com/lowRISC/lowrisc-private-ci/blob/ibex-regression/jobs-ibex-regression.yml

imphil, 2021-01-07:
@udinator FYI, I'll take care of the setup, I only need your input for the dependencies.

GregAC, 2021-12-10:
We now have regular full regression runs.

closed 2021-12-10 by GregAC