lowRISC/ibex #1223: Spike timeouts in riscv_pmp_full_random_test with experimental-maxperf-pmp-bmfull-icache config

imphil, 2021-01-07:
I see spike timeouts in DV runs of the test `riscv_pmp_full_random_test` test. Below is one example, but I see them in multiple runs (not all of them):

```
2021-01-07T15:23:14.2445665Z Thu, 07 Jan 2021 15:23:14 INFO     Running spike sim: out/seed-23315/instr_gen/asm_tests/riscv_pmp_full_random_test_11.o
2021-01-07T15:23:24.2463257Z Thu, 07 Jan 2021 15:23:24 ERROR    Timeout[10s]: /nas/lowrisc/tools/spike-ibex/20201023-git-255bf1cacc599b1413438c269100f3ecd0eb3352/bin/spike --log-commits --isa=rv32imcb -l out/seed-23315/instr_gen/asm_tests/riscv_pmp_full_random_test_11.o &> out/seed-23315/instr_gen/spike_sim/riscv_pmp_full_random_test.11.log
```

From the spike log: The last "normal" instruction is the illegal instruction (is that intentional?). After that spike loops until the timeout through three instructions: j, addi, and sw.

```
core   0: 0xffffffff800018a8 (0x0000ffe2) c.fswsp fs8, 252(sp)
core   0: exception trap_illegal_instruction, epc 0xffffffff800018a8
core   0:           tval 0x0000000000000000
core   0: 0x0000000080001800 (0x0800006f) j       pc + 0x80
3 0x80001800 (0x0800006f)
core   0: 0xffffffff80001880 (0xf8498993) addi    s3, s3, -124
3 0x80001880 (0xf8498993) x19 0x7fffffec
core   0: 0xffffffff80001884 (0x0019a223) sw      ra, 4(s3)
core   0: exception trap_store_access_fault, epc 0xffffffff80001884
core   0:           tval 0x000000007ffffff0
```

Logs:
* stdout log: [log.txt](https://github.com/lowRISC/ibex/files/5782467/log.txt)
* Full build artifacts: https://artproduks1.artifacts.visualstudio.com/Ace3fc167-bb91-4c33-a65a-ae648df7f974/4efdbdf8-a184-4518-9864-57d2a49c2aed/_apis/artifact/cGlwZWxpbmVhcnRpZmFjdDovL2xvd3Jpc2MvcHJvamVjdElkLzRlZmRiZGY4LWExODQtNDUxOC05ODY0LTU3ZDJhNDljMmFlZC9idWlsZElkLzMwMzg0L2FydGlmYWN0TmFtZS9vdXQ1/content?format=file&subPath=%2Fout.tar.xz

Config:
* Config: `experimental-maxperf-pmp-bmfull-icache`
* Seed: 23315
* Tool: VCS 2020.1

@udinator would you mind having a look?

udinator, 2021-01-07:
Hmm, that is a bit strange...not sure why a floating point instruction is being created here.
I'll take a look at this.

udinator, 2021-01-07:
@imphil, would you mind pasting the entire Spike failure log in a txt file?

imphil, 2021-01-08:
@udinator The entire log file is around 60 MB and I cannot attach it to this issue due to its size. You can find it (together with all other log files) in the build artifacts linked in the initial comment at `out/seed-23315/instr_gen/spike_sim/riscv_pmp_full_random_test.11.log` in the archive. Can you access the build artifact link? Otherwise I can copy it over to a Google drive or so.

> is the RTL simulation indicating a test success even though Spike times out?

I'm not quite sure how to interpret the stdout log file (also linked in the initial comment). It seems to ERROR and move on for all runs except for the last run 29, when it ends with "ERROR return code: True/255". Not sure if that's a sign of a different problem or just the "error summary" resulting from the previous problems.

udinator, 2021-01-08:
Ahh yes I am able to access the build artifacts.
I've confirmed my suspicions as to what is causing the Spike timeouts, I'll work on fixing this.
Note that even though Spike will timeout, the RTL simulation should still be passing (unless CI blows up as soon as it sees an ISS simulation error, which might be the case since the stdout log doesn't show anything happening past the ISS stage).

imphil, 2021-03-11:
This test is still timing out in current master (70c3702421feccc0774a9a948aebe94cfa289623).

Here's a more recent run:

```
make -C dv/uvm/core_ibex ISS=spike IBEX_CONFIG=experimental-maxperf-pmp-bmfull
```

```
2021-03-11T10:16:40.4119061Z Thu, 11 Mar 2021 10:16:40 INFO     Running spike sim: out/seed-15769/instr_gen/asm_test/riscv_pmp_full_random_test_13.o
2021-03-11T10:16:50.4136514Z Thu, 11 Mar 2021 10:16:50 ERROR    Timeout[10s]: /nas/lowrisc/tools/spike-ibex/20201023-git-255bf1cacc599b1413438c269100f3ecd0eb3352/bin/spike --log-commits --isa=rv32imcb -l out/seed-15769/instr_gen/asm_test/riscv_pmp_full_random_test_13.o &> out/seed-15769/instr_gen/spike_sim/riscv_pmp_full_random_test.13.log
```

Log files and artifacts:
* [stdout log](https://dev.azure.com/lowrisc/4efdbdf8-a184-4518-9864-57d2a49c2aed/_apis/build/builds/35752/logs/12)
* [Full build artifacts](https://artproduks1.artifacts.visualstudio.com/Ace3fc167-bb91-4c33-a65a-ae648df7f974/4efdbdf8-a184-4518-9864-57d2a49c2aed/_apis/artifact/cGlwZWxpbmVhcnRpZmFjdDovL2xvd3Jpc2MvcHJvamVjdElkLzRlZmRiZGY4LWExODQtNDUxOC05ODY0LTU3ZDJhNDljMmFlZC9idWlsZElkLzM1NzUyL2FydGlmYWN0TmFtZS9vdXQ1/content?format=file&subPath=%2Fout.tar.xz)

@udinator would you mind having a look (or disable the test if we cannot fix it in the near future)? This is blocking running the Ibex DV regression more regularly.

udinator, 2021-03-13:
@imphil, I just created a PR to temporarily remove this from the regression list until I can get some time to fix the issues.