lowRISC/ibex #1228: [tracking] Activate pseudo zext.[b|h] decoding

towoe, 2021-01-09:
The tracer currently does not decode the pseudo instruction.
Use it once it is supported by more tools. Actually not quite which, but it was recently mentioned in https://github.com/llvm/llvm-project/commit/249d7de1190f50178181d2477aa661cd252e294c

vogelpi, 2021-11-26:
I just noted that more recent development versions of the tools we use (e.g. the Spike version for @GregAC 's cosim framework) as well as GCC v.10.2.0 (see https://github.com/riscv-collab/riscv-gcc/tree/riscv-gcc-10.2.0-rvb) and Binutils (https://github.com/riscv-collab/riscv-binutils-gdb) needed for Bitmanip support `zext.h` (`zext.b` is not part of Bitmanip v.1.0.0). So we could change our tracer to output `zext.h`, WDYT? @luismarques do you maybe know what the status in LLVM is on this?

luismarques, 2021-11-26:
> @luismarques do you maybe know what the status in LLVM is on this?

Both the current toolchain and the under evaluation one (based on LLVM 13) support those two instructions (with the appropriate command-line options). I did some quick testing and the encodings didn't seem to have changed. I can double-check that against the specs, if needed.

vogelpi, 2021-11-26:
Thanks for your feedback @luismarques ! This sounds great, I suggest that once we provide both an LLVM and GCC toolchain that support these pseudo-instructions, we make the tracer to output them. I think the encoding is still the same, no need to check.