lowRISC/ibex #1230: Ibex Simple system python requirements concerns

rahulraveendran15-coder, 2021-01-11:
Hi,
IN the link "https://github.com/lowRISC/ibex/tree/master/examples/simple_system" it written in order to get the python requirements need to run the command 

`pip3 install -U python3-requirements.txt
`

I have the following below concerns:

1. The file name mentioned here is wrong, in root repository only python-requirements.txt is present.
2. If we run the above command with correct file name the error will generate cannot find the package.
3. If I run the command with -r then its working. But I have a question what is the use of -U switch there in the above command ?
4. Because I am observing different results when running with -U and -Ur as below:

```
pip3 install -Ur python-requirements.txt

pip3 install -r python-requirements.txt
```
Could you please rectify it and if its true can you please update the page

imphil, 2021-01-11:
It should be `pip3 install -U -r python-requirements.txt`. Would you mind filing a pull request to fix that?

rahulraveendran15-coder, 2021-01-11:
How can I do pull request?
Also what is the use of -U along with pip3 I am not getting its use

imphil, 2021-01-11:
-U is a shorthand for --upgrade, it ensures that all packages are at their latest version. You can find more information about that in the pip3 documentation, and with "pip3 --help".

You can find information on how to contribute to this repository ("do a pull request") in various tutorials on the internet, including the one from OpenTitan at https://docs.opentitan.org/doc/ug/github_notes/.

closed 2021-01-11 by rahulraveendran15-coder