lowRISC/ibex #1236: riscv_arithmetic_basic_test fails for all the iterations

ishita71, 2021-01-12:
Hi,

[compile.log](https://github.com/lowRISC/ibex/files/5799927/compile.log)
[sim.log](https://github.com/lowRISC/ibex/files/5799928/sim.log)

## Observed Behavior
I am trying to run riscv_aritmetic_basic_test in core_ibex  in the master branch but after running, all the test gets fails and shows the following message.
[image]
-->

## Steps to reproduce the issue

towoe, 2021-01-12:
Hi @ishita71, in your sim.log file I see the following message:
```
  Error-[UCLI-LOADDL-INTERNAL] loaddl command Internal error.
    Internal error in loaddl command : Load.
    /global/apps/verdi_2020.03/verdi/Q-2020.03/share/PLI/VCS/LINUXAMD64/libnovas.so:
    cannot open shared object file: No such file or directory
    Please contact vcs_support@synopsys.com or call 1-800-VERILOG.
  
  Please make sure VERDI_HOME is properly set, and rerun the simulation. For 
  more details, please refer the FSDB user guide for right usage.
```
So it might be a tool / installation issue.
Could you please verify that VCS is working properly in your setup?

ishita71, 2021-01-12:
Thanks for helping out !

After fixing VERDI_HOME issue , testcase still gets failed.

[image]

I have attached sim.log ,compile.log and regr.log file

[sim.log](https://github.com/lowRISC/ibex/files/5801810/sim.log)
[compile.log](https://github.com/lowRISC/ibex/files/5801816/compile.log)
[regr.log](https://github.com/lowRISC/ibex/files/5801807/regr.log)

Can you please help me with this?

 Regards,
Ishita Shah

towoe, 2021-01-12:
Hi Ishita,
looks like a problem with spike.
'regr.log' file shows that Ibex produced a trace, but I think spike might not.
Please check 'spike : out/seed-31085/instr_gen/spike_sim/riscv_arithmetic_basic_test.0.csv'.
And in the screenshot the error is reported as a timeout when calling spike. So my guess would be that there is a problem with that. You might want to check the log output created there.

rswarbrick, 2021-01-25:
Hi Ishita. Have you managed to look at this further? Since these tests pass locally, I suspect it's a tool version issue.

dawidzim, 2021-02-02:
Hey!
Try to run with LowRISC Spike version:
https://github.com/lowRISC/riscv-isa-sim/tree/ibex

rswarbrick, 2021-02-02:
Right, I don't think there's anything more to do at this end. @ishita71, please re-open if there's something still broken.

closed 2021-02-02 by rswarbrick