lowRISC/ibex #1237: Floating Point Unit (FPU) integration with ibex

zeeshanrafique23, 2021-01-13:
I am planing to integrate a parameterized floating point unit that is design by [PULP-Plateform](https://github.com/pulp-platform/fpnew) with ibex, so can any one tell me it will be a good part time project to get hands-on on RISC-V F, D and Q extension.
And is this core is good to support FPU for above extension?

GregAC, 2021-01-14:
I'm not hugely familiar with F, D and Q extensions but getting hands on with an FPU and integrating it with a RISC-V cores sounds like a decent way to build some knowledge of them.

Ibex is a simple core so you shouldn't have much trouble adding new instructions, though it's also 32-bit. So you may encounter extra issues with implementing D and Q as it won't easily fit with existing data paths.

zeeshanrafique23, 2021-01-14:
Yes, you are right actually there are certain things that need to be changed accordingly while implementing FPU, I think if there will be a separate datapath for D, and Q extensions so that would not make any hurdles any more and those paths must be muxed based on instructions catered inside the core.

Thanks for your input, I will make this issue open to have a few more point of views.

closed 2021-02-17 by zeeshanrafique23