lowRISC/ibex #1244: Not able to run hello model with rv32emc

rahulraveendran15-coder, 2021-01-21:
Hi,
I am trying to run hello software program with RV32EMC instruction set. On compilation of the files it is showing me the following error. How can I run hello_world  with RV32EMC architecture? It is showing error from crt0.S file

[image]

GregAC, 2021-01-21:
The 'E' extension cuts down the number of registers to 16. crt0.S contains instructions which use registers above `x15` so won't work with the `E` extension.

We don't have any example software designed for use with the `E` extension and I'm uncertain how good the compiler support for it is. You'd need to write some and experiment for yourself.

rswarbrick, 2021-01-25:
I'm closing this issue as I believe the question has been answered. Please re-open or file a new issue if there's something we've missed.

closed 2021-01-25 by rswarbrick