lowRISC/ibex #1248: Report a bug in Ibex

Imperas, 2021-01-24:
I note in the doc: https://github.com/lowRISC/ibex/blob/master/doc/03_reference/verification.rst you say:

> The master branch of Spike and OVPSim may support a different version. It is recommended the lowRISC-specific branch of Spike is used when using a configuration with Bitmanip to ensure the simulated version of the Bitmanip specification matches with the RTL implemented version.

This is not giving the correct picture - yes you can use OVPsim with the latest version and you get the current spec version as 'master' [for today that is bitmanip 0.94-draft] but as you correctly indicate that is pretty useless for verification where you are trying to verify another version of the spec, eg v0.92.

OVPsim has parameters to set the  specific version of the individual ISA extensions that you are using - so if it is bitmanip, crypto, vector, debug, p-dsp, user, priv, etc., - whatever - the simulator allows specific control - so you get what you are going to silicon with - users say this is a far better approach than each of them having to maintain their own modified branches of simulators for their silicon - they make use of simulators (like OVPsim)  that are focused on helping people be more effective and efficient with their verification.

So maybe you could clarify in the doc the OVPsim setting of spec versions and not just being the head like spike..

Also, if you want help setting the correct spec versions for OVPsim - let us know how we can help.

Thanks
Simon
Imperas Software (creators of OVPsim)

rswarbrick, 2021-01-25:
Thanks for the report. I'll happily fix this, but want to make sure that I describe things correctly. Glancing quickly through https://github.com/riscv-ovpsim/imperas-riscv-tests/tree/main/riscv-ovpsim, it looks like support for specific spec versions is only available in the commercial version. Is that correct?

Imperas, 2021-01-25:
Glad you will update things. The situation is:
The free GitHub riscvOVPsim supports the ratified specs and those close to
ratification. If the simulator supports an ISA extension, then its version
is controllable.
The free OVPworld.org riscvOVPsimPlus, requires registration to download,
and supports more ISA extensions and includes those under development like
Zfinx, P-DSP when they are available etc., and again if the simulator
supports the extension, then it is version controllable.

Thanks
Simon

-- 
====================================================================
The information contained in this electronic mail message and any
attachments hereto
is privileged and confidential information intended only for the use of the
individual or
entity named above or their designee.  If the reader of this message is not
the intended
recipient, you are hereby notified that any dissemination, distribution or
copying of this
communication is strictly prohibited.  If you have received this
communication in error
please immediately notify us by return  message or by telephone and delete
the
original message from your mail system.  Thank you.
====================================================================

PR #1252 (2021-01-25): [doc] Update dependency descriptions for Spike/OVPsim
This should match what's going on a bit more accurately. The link to
OVPsim now points at the (free of cost) commercial tool: riscv-ovpsim
doesn't support the bitmanip specification that we're using at the
moment.

Fixes #1248.
closed 2021-01-25 by rswarbrick