lowRISC/ibex #1249: Inline assembly syntax for adding custom instructions to Ibex.

subhajit26, 2021-01-25:
Hi,
        Can I get an idea of what inline assembly syntax to be followed for adding custom instructions to Ibex ? Is it the one that is mentioned in OpenTitan document (Extended ASM) ?

rswarbrick, 2021-01-25:
Hi there. If want to add a custom instruction to Ibex, there are several things you'll need to do:

1. Write some snippets of assembly to try it out. The RISC-V assembler in Binutils has a rather cryptic [.insn](https://sourceware.org/binutils/docs/as/RISC_002dV_002dDirectives.html) directive that you can use to write them (or you could even start by assembling the code in your head and just writing hex constants)
2. Add support for the instruction to the RTL
3. Add support for the instruction to an instruction set simulator like Spike (otherwise any testing is going to be painful)

I think that the "extended ASM" that you mention is what is used in C compilers like GCC for embedding snippets of assembly code into C.

subhajit26, 2021-01-25:
Hi,
1. We are going to be using  Imperas as ISS for our thesis work. So please leave any suggestion as I can see Imperas as one of the ISS for Ibex.
2. Is it possible to add new instruction to the ibex as you mentioned in point 1 without adding it to the C code and compiling it? If it is possible , can you give some suggestion.
3. For last 2 days I have been exploring how to add the custom instructions and all I have learnt is that we need to add some inline assembly syntax in C code . Please suggest. Even a video of Imperas  says that.
4. In openTitan document I can see something as builtin function. Can you please tell their use in regards to the addition of custom instructions.
All your suggestion will be highly appreciated since I am very much new to this custom instruction process.

subhajit26, 2021-01-25:
Hi, 
    I am bit confused about the point 2 of your suggestion. By "adding support to RTL" do you mean to say that the decode stage should be able to decode the opcode and recognize the instruction or it is something else that I am missing. Please leave your suggestion.

rswarbrick, 2021-01-25:
Re using Imperas (presumably the proprietary version of OVPSim), there's some discussion in issue #1248.

Your other questions suggest that you probably need to do a bit of reading about how processors work and how e.g. C code gets compiled then how assembly code is assembled, decoded and executed. This isn't really an Ibex-specific thing, and I don't think GitHub issues are the best place to discuss it. If you need more help, the best place to start is probably the lowRISC Zulip channel. See https://www.lowrisc.org/community/ for details of how to join.

rswarbrick, 2021-04-05:
Hi there! I'm closing this now, since the discussion seems to have petered out. Please feel free to re-open if you would like to ask further specific questions.

closed 2021-04-05 by rswarbrick