lowRISC/ibex #1253: Yosys synthesis error

rahulraveendran15-coder, 2021-01-25:
Hi,
While running synthesis with Yosys, it is throwing error from the file "ibex_cs_registers.v"

10. Executing Verilog-2005 frontend: syn_out/ibex_25_01_2021_15_08_46/generated/ibex_cs_registers.v
syn_out/ibex_25_01_2021_15_08_46/generated/ibex_cs_registers.v:538: ERROR: Non-constant function call in constant expression.
Is there any way to fix this

rswarbrick, 2021-01-25:
As usual, can you give the exact command that you're running? I know that @GregAC has been doing some Yosys-based synthesis recently, so I'd expect this to work.

GregAC, 2021-01-25:
There is a small issue with synthesis on current master which I fix here: https://github.com/lowRISC/ibex/pull/1255 however I don't think this is the same problem.

With the fix from that PR I can successfully run synthesis on my machine. What versions of sv2v and yosys are you running? yosys v0.9 should work and I am currently using sv2v version: `v0.0.5-18-ga170536`

rahulraveendran15-coder, 2021-01-26:
@GregAC 
I am using Yosys version Yosys 0.9+3858 (git sha1 410ea422, clang 3.4.2 -fPIC -Os).
How can I check sv2v version? I suspect the error is from sv2v

rahulraveendran15-coder, 2021-01-26:
@GregAC 
I am using yosys 0.9 and sv2v v0.0.6-9-gb28a3ca?
Can this be the cause?

GregAC, 2021-01-26:
It seems likely it's an sv2v issue (or potentially a yosys issue, sv2v emitting correct verilog that yosys isn't accepting). I would suggest trying another version of sv2v, e.g. the 0.0.5 tag https://github.com/zachjs/sv2v/releases/tag/v0.0.5

rahulraveendran15-coder, 2021-01-26:
@GregAC 
I cloed the specific tag v0.0.5 version of sv2v from git hub. But still the same error is coming.
The commands I used for sv2v  make is as below:
 
```
  git clone https://github.com/zachjs/sv2v.git
 cd sv2v
 git checkout tags/v0.0.5
 make

```

Any hint on this, after this the version of sv2v is showing as `sv2v v0.0.5`, will it be a yosys issue? But we both are using same version of yosys right? In the below image you can see 10 th one is throwing error.

[image]

rahulraveendran15-coder, 2021-01-26:
My Yosys full version is,

`Yosys 0.9+3858 (git sha1 410ea422, clang 3.4.2 -fPIC -Os)`

rswarbrick, 2021-01-26:
Looks like this is caused by the Yosys version. I've just reproduced what you're seeing and will have a proper look at what's going on later today.

rahulraveendran15-coder, 2021-01-26:
@rswarbrick 
Thanks for the update. Basically what I understood is that Yosys added a new syntheisable construct in their latest update which is causing the issue.

rswarbrick, 2021-01-26:
Bisecting Yosys builds points at commit 2085d9a55 (committed on 1st Jan 2021) being the first version that fails. I'll try to understand what's going on and then follow up on the Yosys bug tracker.

rswarbrick, 2021-01-27:
Right, looks like the Yosys guys are working on a fix. I don't think there's anything else to do here (the next Yosys release will probably work again).

closed 2021-01-27 by rswarbrick