lowRISC/ibex #1254: Uvm package not found

ThibaultRicher, 2021-01-25:
It seems like I don't have the uvm-pkg in my directory and when I compile your core_ibex_tb_top.sv

## My Environment

I used Ibex to simulate it functionnaly make a synthesis of it and after doing the place&route in order o use it in an ASIC's project

**EDA tool and version:**

QuastaSim-64 vlog10.7a_1 Compiler 2018.05

**Operating system:**

RedHat7.9

**Version of the Ibex source code:**

The last one

rswarbrick, 2021-01-25:
Please can you give the exact command that you ran? This sounds a lot like you're not running fusesoc, but it's hard to tell what's going on without more details.

ThibaultRicher, 2021-01-26:
[core_ibex_tb_top.compilation.csh.docx](https://github.com/lowRISC/ibex/files/5872622/core_ibex_tb_top.compilation.csh.docx)
This is it sorry for the response time, I had to get my csh out of my Redhat computer.

rswarbrick, 2021-01-26:
If you're writing your own flow like this, I'm afraid you're kind of on your own.

Squinting at your document, it looks like you're trying to run something a bit like the UVM flow in `dv/uvm/core_ibex`. That code is supposed to support Questa: I suggest you use that instead of rolling your own.

I'm going to close this issue, because it sounds like the problem is that you're having trouble finding your UVM installation when running Questa with your own scripts: a bit hard for us to fix! But there are likely to be rough corners with the code in `dv/uvm/core_ibex` when using Questa (we don't have a Questa licence, so haven't run that recently). Please try following the instructions at https://ibex-core.readthedocs.io/en/latest/03_reference/verification.html and, if something doesn't work, open an issue about that.

closed 2021-01-26 by rswarbrick