lowRISC/ibex #1263: Doubt in .dis file generated under examples/sw/

rahulraveendran15-coder, 2021-01-29:
Hi,
I have a doubt in the .dis file generated in the example/sw/simple_system.
In the .dis file generated there above every pc count I am seeing one more instruction, how is it generated and what is the purpose of it?

IN the below snapshot if you see  the bld instructions. on top of some instructions its not getting generated (1006e0 & 1006e4)

  **addi x10, x0, 0**
  1006d8:	4501                	li	a0,0
  
  **addi x11, x0, 0**
  1006da:	4581                	li	a1,0
  
  **jal x1, main**
  1006dc:	00a030ef          	jal	ra,1036e6 <main>
  
  **li x5, SIM_CTRL_BASE + SIM_CTRL_CTRL**
  1006e0:	000202b7          	lui	t0,0x20
  1006e4:	02a1                	addi	t0,t0,8
  
  **li x6, 1**
  1006e6:	4305                	li	t1,1

  **sw x6, 0(x5)**
  1006e8:	0062a023          	sw	t1,0(t0) # 20000 <_stack_len+0x18000>

[image]

rswarbrick, 2021-02-01:
This is caused by the fact that we're generating the disassembly listing with the `-S` argument to objdump, which interleaves source code with disassembly. Normally, you'd get hunks of C here (look at another function like `timer_enable`), but `main_entry` is written in assembly so you get this rather silly "double listing".

For extra confusion, the disassembler is also generating pseudo-operations: `li a0, 0` is a synonym for `addi x10, x0, 0` (once you do the ABI renaming too).

Honestly, I don't see the point in us running objdump like this as part of the build: someone who actually needs an assembly listing can probably do it themselves, and will get much less confused by things like this happening if they know what flags they passed. I'll file a PR in a minute to get rid of the step entirely.

rahulraveendran15-coder, 2021-02-03:
Thank you. 
Closing the issue

PR #1265 (2021-03-02): Don't automatically generate disassembly in the example Makefile
Anyone who needs to disassemble their generated ELF can probably just
call objdump directly and the precise set of flags have already
confused at least one potential contributor[1].

We're keeping the canned objdump command for "engineers that know
where to look" because some have said they find it useful.

[1] https://github.com/lowRISC/ibex/issues/1263

Fixes #1263 (marking as a fix because the question becomes moot!)
closed 2021-02-03 by rahulraveendran15-coder