lowRISC/ibex #1267: Ibex Compiler Change

sevvalizmirli, 2021-02-03:
I want to use different compiler. Can I change Gcc with another compiler and how can I change it ?

rswarbrick, 2021-02-03:
@luismarques, who I think knows the most about our RISC-V Clang/LLVM toolchain.

You'll probably have to explain *why* you need a different compiler for us to help you more.

sevvalizmirli, 2021-02-03:
We are students who want to do academic research from TOBB University, Turkey. We want to observe CPU Performance  with some compiler changing. We will use LLVM instead of GCC. We are using ibex for our research. However I could not find how to change GCC with LLVM.

luismarques, 2021-02-10:
Hi @sevvalizmirli.

Could you please describe what is your current workflow for benchmarking Ibex using GCC? Without that information I'm not quite sure how to best help you. Still, here's my best attempt.

In our OpenTitan project (which uses Ibex) we have tools to set up the device firmware compilation with either GCC or LLVM/Clang toolchains. See [1] for an overview. Outside of OpenTitan, my own use of Ibex with Clang was always set up from scratch, so I didn't modify any build system or similar tool that the Ibex repo might contain or use. You can browse around and experiment with the OpenTitan repository for details, but in general I would say that what you want is to use the `clang` tool instead of `gcc` with the correct option for RISC-V. That might be `--target` (e.g. `target=riscv32`), as well as the usual `-march` and `-mabi` options. The target can be automatically recognized if you prefix your compiler binaries with the target triple, e.g. `riscv32-unknown-elf-clang`. Unless you have a complete Clang/LLVM toolchain with the appropriate libraries (e.g. Compiler-RT) and tools (e.g. LLD), you'll probably end up depending on a companion GNU toolchain. For Clang to use that GCC installation they need to either be appropriately installed side-by-side or for you to explicitly specify the `--gcc-toolchain=<your GCC install path prefix>` option. One tricky thing is that the `--gcc-toolchain` option will silently fail if the target triple doesn't match. You can use the `-v` option to check that `clang` is correctly detecting the GCC installation. You might also need the `--sysroot=<some path>` option for Clang to use the system headers and libraries from the GNU toolchain. In some cases, you might also have to change some compilation flags and linker script options for everything to work or to perform at its best, but that probably won't be necessary for you to get started.

Let me know if that helps.

[1] https://docs.opentitan.org/doc/ug/install_instructions/

rswarbrick, 2021-04-05:
Hi there! I'm closing this because there's not been a response for quite a while. @sevvalizmirli: Please feel free to re-open if you have further questions, but it would be good if you could start by responding to Luis, who looks like he put quite a lot of work into replying to you here.

closed 2021-04-05 by rswarbrick

Eymay, 2023-04-04:
Hi @luismarques,
We are modifying and building LLVM and want to see it in action with Ibex. I've modified the provided Makefile in Simple System example.
```
ARCH = rv32i 
TARGET = riscv32
# ARCH = rv32im # to disable compressed instructions
SRCS = main.c encrypt.c round.c decrypt.c
#SRCS = main.c encrypt.c roundInline.c decrypt.c

CC = PATHTOLOWRISCTOOLCHAIN/lowrisc-toolchain-rv32imcb-20230228-1/bin/riscv32-unknown-elf-gcc
CLANG = PATHTOOURLLVMBIN/clang

OBJCOPY ?= $(subst gcc,objcopy,$(wordlist 1,1,$(CC)))
OBJDUMP ?= $(subst gcc,objdump,$(wordlist 1,1,$(CC)))

LINKER_SCRIPT ?= link.ld
CRT ?= crt0.S
CFLAGS ?= --target=$(TARGET) -march=$(ARCH) -mabi=ilp32 -static -mcmodel=medany \
	-fvisibility=hidden -nostdlib -nostartfiles -lgcc --gcc-toolchain=$(CC) $(PROGRAM_CFLAGS)
```

When we use the .o producing rule with `$(CLANG)` instead of GCC these errors occur:
`fatal error: 'stdio.h' file not found
#include <stdio.h>
`

So I guess as you said Clang should be side by side with GCC as GCC can find stdio but Clang cannot. To have that I tried adding `-lgcc` and `--gcc-toolchain`.

How can we achieve to include standard libraries with Ibex defaults like `-nostartfiles` probably preventing it?