lowRISC/ibex #1268: JAL instruction behaviour on Ibex

rahulraveendran15-coder, 2021-02-04:
Hi,
I have a doubt on the behaviour of JAL Instruction.
In the below line :
```
  100a32:	e76ff0ef          	jal	ra,1000a8 <puts>
  100a36:	0001                	nop
```

1. The PC will jump to the puts function written in 1000a8 and the return back address  "100a36" is stored in `ra`

But in some places of the disassembly file I am seeing the `jal` instruction only with offset and no `ra`.
How this will work without return address can anyone please help me on this. Please correct if I am wrong on my first point

Snapshot is attached below:

[image]

rswarbrick, 2021-02-04:
This is a pseudo-instruction version of JAL (which writes the link address to the zeros register). RISC-V assembly has all sorts of helpful pseudo-instructions which, frankly, just make life confusing.

rahulraveendran15-coder, 2021-02-04:
@rswarbrick 
Sorry to ask again, I didnt get your point (which writes the link address to the zeros register)), if the link address is in zeros register how the function will return back?

As you can see in my snippet `jal  1000e2` , calling sum function at pc `1000d4`. I think the link address you are pointing to is the address to which pc should jump. If  it writes the link address to the zeros register there shouldn't be any jump right?
In other case if you are telling the rd register is pointing to zeros register, in this case then how it will return back to puts function after executing the sum function  from 1000d4 and to execute lines below it (which is in puts)

rswarbrick, 2021-02-04:
The C compiler looks like it has generated a [tail call](https://en.wikipedia.org/wiki/Tail_call) here.

zeeshanrafique23, 2021-02-17:
@rahulraveendran15-coder 
There are two conditions for `jal` instruction in RVC (compressed):
1. If the destination register in instruction encoding is x1 or ra `(exp: jal ra <label>)` -> C.JAL
2. If the destination register in instruction encoding is x0 `(exp: jal x0 <label>)` -> C.J

The `jal 100e2 <sum>` instruction in your code is a compressed jump instruction(16-bit wide), and remember that the compressed  `jal` instructions do not encode the destination register's address(rd), it is the responsibility of the assembler to make it C.J or C.JAL according to the condition and in the processor side there is should be complete decoding that differs the jal instructions on the basis of "func3" bits [15:13].

So, according to " 'Volume 1: RISC-V Unprivileged ISA V20191213' section 16.4":
```
C.J performs an unconditional control transfer. The offset is sign-extended and added to the pc to
form the jump target address. C.J can therefore target a ±2 KiB range. C.J expands to jal x0,
offset[11:1].

C.JAL is an RV32C-only instruction that performs the same operation as C.J, but additionally
writes the address of the instruction following the jump (pc+2) to the link register, x1. C.JAL
expands to jal x1, offset[11:1].
```
And @rswarbrick I think it's not a trail call here, the instruction `j 1000ca<puts+0x22>` is used here to skip the subroutine `putschar(*str++)` and jumps to `while (*str)`, hence no need to save the return address because the bnez instruction at 1000d2: will decide either jump to  `putschar(*str++)` or move forward.

Note: Link address means PC+4 or PC+2 depending upon the instruction.
The x0 use as a destination when you don't want to store the return address (your purpose is only to jump) most of the time used as a trail call.

rswarbrick, 2021-04-05:
I think this question has been answered (very thoroughly! Thanks @zeeshanrafique23!) Please feel free to re-open if you think there's something that's still unclear.

closed 2021-04-05 by rswarbrick