lowRISC/ibex #1272: Doubts regarding multiply wait and divide wait:

subhajit26, 2021-02-12:
Hi,
     Please see the code below where I am using no division at all but still it is showing me divide wait : Can you please explain to me the concept divide wait and the multiply wait. 
```
#include <stdio.h>
#include "simple_system_common.h"

int mul(int a, int b) 
{   
    if (a == 0 || b == 0)
        return 0;
 
    if (b == 1)
        return a;
 
    if (a == 1)
        return b;
 
    return a + mul(a, b - 1);
} 

int main() 
{ 
    
    static const int array[10] = {1,2,3,4,5,6,7,8,9,10};
    
    int i;
    int sum=0;
    for(i=0; i < 9 ; i++){
        sum+= mul(array[i],array[i+1]);
       
        puts("sum=");
        putdec(sum);
        puts("\n");
    }
    
    return 0; 
} 
```
This is the result after running the ./ build command.
```
Cycles:                     8835
NONE:                       0
Instructions Retired:       4367
LSU Busy:                   1980
Fetch Wait:                 440
Loads:                      1272
Stores:                     708
Jumps:                      348
Conditional Branches:       361
Taken Conditional Branches: 260
Compressed Instructions:    2343
Multiply Wait:              0
Divide Wait:                1440

```

subhajit26, 2021-02-12:
Hi,
       I also tried the same code * operator but then the divide wait is still same  which is bit confusing to me. Please see the code:
```
#include <stdio.h>
#include "simple_system_common.h"

int multiply(int n, int m) 
{   
    int ans = 0;
    ans = n*m; 
    return ans; 
} 
  
// Driver code 
int main() 
{ 
     
    static const int array[10] = {1,2,3,4,5,6,7,8,9,10};
    
    int i;
    int sum=0;
    for(i=0; i < 9 ; i++){
        sum+= multiply(array[i],array[i+1]);
        puts("sum=");
        putdec(sum);
        puts("\n");
        
    }
     
    return 0; 
} 
```
This is the result after running the ./ build command.
```
Cycles:                     6571
NONE:                       0
Instructions Retired:       3043
LSU Busy:                   1398
Fetch Wait:                 303
Loads:                      857
Stores:                     541
Jumps:                      251
Conditional Branches:       157
Taken Conditional Branches: 118
Compressed Instructions:    1496
Multiply Wait:              18
Divide Wait:                1440
```

GregAC, 2021-02-12:
How's `putdec` implemented? Generally this will involve use of modulus and divide  (e.g. %10 to get bottom decimal digit, then / 10 to shift it off the bottom to look at the next). This is a likely source of the divide wait.

Note you can enable and disable counters to look at regions of interest only, see the `pcount_enable` function in https://github.com/lowRISC/ibex/blob/master/examples/sw/simple_system/common/simple_system_common.h

subhajit26, 2021-02-12:
Hi Greg,
                Thank you very much for pointing out the particular place. Yes, putdec is the source of divide wait.

rswarbrick, 2021-04-05:
Hi there! Looks like Greg's response resolved this back in February so I'm marking this as closed.

closed 2021-04-05 by rswarbrick