lowRISC/ibex #1273: Adding custom instruction support to the hardware.

subhajit26, 2021-02-15:
Hi,
       I have implemented 2 custom instructions 1. mod 2. mac and I have also added support for them in the spike which seems to be working fine. Now I want to add the support in hardware . As per my understanding till now I believe adding to the files ibex_pkg.sv, ibex_decoder.sv and ibex_multdiv_fast.sv will be enough. Please suggest if there is any more file that I need to consider and also suggest if there is anything more that I need to consider for adding the support to the Ibex.

subhajit26, 2021-02-17:
Hi,
    I am trying a custom instruction and want to add the support in hardware to make it work.
```
int main() 
{ 
     
    static const int array[10] = {1,2,3,4,5,6,7,8,9,10};
    
    int i;
    int sum=0;
    for(i=0; i < 9 ; i++){

    asm volatile
   (
    "mac   %[z], %[x], %[y]\n\t"
    : [z] "+r" (sum)
    : [x] "r" (array[i]), [y] "r" (array[i+1])
   );  
        //sum+= multiply(array[i],array[i+1]);

        puts("sum=");
        putdec(sum);
        puts("\n");
   }
     
    return 0; 
} 
```
The instruction support has been added in both tool-chain and ISS (Spike). As far as the hardware is concerned I have modified two files:
1. <ibex_pkg.sv>: I have defined instruction pattern in such a way that if fall under OPCODE_OP category as  the instruction class in M.
```
typedef enum logic [2:0] {
  // Multiplier/divider/MAC
  MD_OP_MULL,
  MD_OP_MULH,
  MD_OP_DIV,
  MD_OP_REM,
  MD_OP_MAC
} md_op_e;

```
2. <ibex_decoder.sv .>
I have added this portion for the unque case(opcode):
```
{7'b000_1000, 3'b111}: begin // mac
              multdiv_operator_o    = MD_OP_MAC;
              multdiv_signed_mode_o = 2'b00;
              illegal_insn          = (RV32M == RV32MNone) ? 1'b1 : 1'b0;
            end
```
I have added this portion for the unique case (opcode_alu):
```
{7'b000_1000, 3'b111}: begin // mac
              alu_operator_o = ALU_ADD;
              mult_sel_o      = (RV32M == RV32MNone) ? 1'b0 : 1'b1;
            end

```
Please suggest whether this is the correct way to add the instruction in decoder module.

Regarding<ibex_multdiv_fast.sv> file I can see mac operation is taking place internally with accumuator and summand signals. I am not being able to find the link that how to connect my instruction to that functionality. Now in disassembly it is showing mul and add instruction but I want this operation to show only mac instruction in disassembly . The instruction pattern of mac differs from other M class instruction in  the 7 bit MSB function field.

GregAC, 2021-02-17:
Hi @subhajit26 that looks like a reasonable way to do things in the decoder, though can't offhand say if it will achieve exactly what you need. Note that for a multiply accumulate you will require a new data path from the multiple unit to the alu to perform the add so further work will be required.

I can't comment on the disassembly side of things. I'm not hugely familiar with the tool internals there plus that's not really an Ibex related issue.

subhajit26, 2021-02-18:
Hi Greg,
              "Note that for a multiply accumulate you will require a new data path from the multiply unit to the alu to perform the add so further work will be required."

Paying heed to what you said I was trying to implement the custom mac instruction only to the fsm for RV32MFast.
```
ALBH: begin
          // al*bh<<16
          mult_op_a = op_a_i[`OP_L];
          mult_op_b = op_b_i[`OP_H];
          sign_a    = 1'b0;
          sign_b    = signed_mode_i[1] & op_b_i[31];
          // result of AL*BL (in imd_val_q_i[0]) always unsigned with no carry
          accum     = {18'b0, imd_val_q_i[0][31:16]};
          //if (operator_i == MD_OP_MULL) begin
          if (operator_i == MD_OP_MAC) begin
            mac_res_d = {2'b0, mac_res[`OP_L], imd_val_q_i[0][`OP_L]};
          end else begin 
            // MD_OP_MULH
            mac_res_d = mac_res;
          end
          mult_state_d = AHBL;
          
        end

        AHBL: begin
          // ah*bl<<16
          mult_op_a = op_a_i[`OP_H];
          mult_op_b = op_b_i[`OP_L];
          sign_a    = signed_mode_i[0] & op_a_i[31];
          sign_b    = 1'b0;
          //if (operator_i == MD_OP_MULL) begin
          if (operator_i == MD_OP_MAC) begin
            accum        = {18'b0, imd_val_q_i[0][31:16]};
            mac_res_d    = {2'b0, mac_res[15:0], imd_val_q_i[0][15:0]};
            mult_valid   = 1'b1;

            // Note no state transition will occur if mult_hold is set
            mult_state_d = ALBL;
            mult_hold    = ~multdiv_ready_id_i;
          end else begin 
            accum        = imd_val_q_i[0];
            mac_res_d    = mac_res;
            mult_state_d = AHBH;
            
          end
        end
```

#include <stdio.h>
#include "simple_system_common.h"

```
int main() 
{ 
     
    int a=3;
    int b=4;
    int sum = 1;

    asm volatile
   (
    "mac   %[z], %[x], %[y]\n\t"
    : [z] "+r" (sum)
    : [x] "r" (a), [y] "r" (b)
   ); 
 
  

        puts("sum=");
        putdec(sum);
        puts("\n");
   
     
    return 0; 
} 
```
My only concern is that my code is giving me the multiplication for all the values but it is not adding up when I am using MD_OP_MAC  whereas when I am using MD_OP_MULL it is giving the result of mac operation.
It seems like MD_OP-MAC is only giving me the mac_res_d: the intermediary value. Can you please point out the line where it is performing the accumulation for RV32MFast  FSM. I am bit confused about how the accumulator is functioning in the FSM.

rswarbrick, 2021-04-05:
Hi there! I'm going to close this because it's been dormant for a while and it's not really clear what we can best do to help you. If you're still working on this and have a more specific question, feel free to re-open or to open a new issue.

closed 2021-04-05 by rswarbrick