lowRISC/ibex #1274: Synthesis with Cadence

nagendragd, 2021-02-15:
## Observed Behavior

I am trying to synthesize ibex_core with Cadence. As Cadence supports SystemVerilog natively, I tried that.

read_hdl -language sv {path to SV file}

That did not work and reported syntax errors in the ibex_core.sv file. Such as:
Error   : Parsing error. [VLOGPT-1] [read_hdl]
        : Expecting identifier in file '../rtl/ibex_core.sv' on line 22, column 31.
        : Invalid Verilog syntax is parsed, or unsupported Verilog syntax is encountered.
    parameter ibex_pkg::rv32m_e   RV32M            = ibex_pkg::RV32MFast,

Then I tried to get around it by converting to Verilog via sv2v, but sv2v reported an error with ibex_tracer.sv.
sv2v: invalid based literal digits: (2,25,268435455,268435455,28,25)
CallStack (from HasCallStack):
  error, called at src/Language/SystemVerilog/AST/Number.hs:239:9 in main:Language.SystemVerilog.AST.Number

I would like to get an example or a reference for how either approach worked out ok for another user - either directly using SV or via sv2v.

## Expected Behavior

## Steps to reproduce the issue

sv2v error:
I cloned the lowRISC git hub repo to my Ubuntu machine
Setup syn_setup.sh per my settings
Ran syn_yosys.sh and saw the error
I did not modify RTL

Cadence SV error:
I cloned the lowRISC git hub repo to my RHEL7 machine
Created my own do_synth.tcl 
Added a read_hdl -language sv ibex_core.sv entry to it
Ran this tcl inside Cadence command line synth tool
Cadence reported parser errors with the ibex_core.sv source file
Again I did not modify RTL.

Hope this helps.

## My Environment

**EDA tool and version:**

**Operating system:**

The Cadence experiment was done on RHEL 7.
The sv2v experiment was done on Ubuntu 18.04.5 LTS

**Version of the Ibex source code:**

I am using the latest master branch version available on the lowRISC github repo.

GregAC, 2021-02-17:
Looks like the Cadence tool you are using doesn't support using enums as a parameter value, you could try replacing it with the integer literal instead (which is 2 in this case, look in `ibex_pkg.sv` to see the various enum values).

This is valid Verilog so we'd prefer not to change this unless needed, though we could consider a workaround if this broken in Cadence tooling. Are you using the latest version of your Cadence tool?

rswarbrick, 2021-04-05:
Hi! I'm closing this, since it's been quite a while since Greg asked a follow-up with no response. Please feel free to re-open if you'd like to work on it again.

closed 2021-04-05 by rswarbrick