lowRISC/ibex #1275: Software model fails while running in Xcelium

rahulraveendran15-coder, 2021-02-18:
Hi,
I am facing a strange behaviour while running the simple system using Xcelium. The same code is running and getting expected output when using verilator.
My C code is as below:'
```
typedef struct { 
    char name[80]; 
    int age; 
    float percentage; 
} statistics_i16_state_t; 

int main(void) {
int val,val_1,a,b;
    char *float_result = "";
a = 2;
b = 3;
val_1 = a + b;
 statistics_i16_state_t  *emp;
emp->age=18;
putdec(emp->age);

}
```

The line emp->age=18; is causing the assertion failure while running in Xcelium. I have attached the xcelium log, disassembly file
[xrun.log](https://github.com/lowRISC/ibex/files/6001948/xrun.log)
[hello_test_dis.log](https://github.com/lowRISC/ibex/files/6001958/hello_test_dis.log)

 for details.

tomroberts-lowrisc, 2021-02-18:
Looks like the simulation is failing due to reading X from memory (which doesn't happen in verilator since it doesn't support X).

The X is likely due to reading uninitialized memory.
```
 statistics_i16_state_t  *emp;
```
Here you create a pointer to a struct but never allocate any memory to it before accessing it. It would be worth reading through some C code tutorials around pointers and memory management.

rahulraveendran15-coder, 2021-02-18:
Thank you for the comments.

closed 2021-02-18 by rahulraveendran15-coder