lowRISC/ibex #1277: PMP behavior when PMP is off and in U-mode

mmxsrup, 2021-02-18:
## Observed Behavior
I found that the behavior is different between spike and ibex when transitioning from m-mode to u-mode with the `mret` instruction without setting pmp. In ibex, after the `mret` instruction, the fetch of the next instruction causes a pmp error. On the other hand, spike continues to operate without causing a pmp error even after the `mret` instruction.
In the implementation of ibex, if pmp is not set (`PMP_MODE_OFF`), `region_match_all` will be zero and u-mode access will unconditionally generate an error. This makes a difference between ibex and spike.  Is it the expected behavior to unconditionally raise a pmp error when pmp is not set?
https://github.com/lowRISC/ibex/blob/65287f7d7b5a9b0733f37c01b292b1de08656e5c/rtl/ibex_pmp.sv#L155-L160

## Steps to reproduce the issue
I overwrote the program to `example/sw/simple_system/hello_test/hello_test.c`.

```c
#include "simple_system_common.h"

#define MSTATUS_MPP_M (3 << 11)
#define MSTATUS_MPP_U (0 << 11)

void test() {
  return;
}

int main(int argc, char **argv) {
  // from M mode to U mode
  uint32_t mstatus;
  asm volatile("csrr %0, mstatus" : "=r"(mstatus));
  mstatus = (mstatus & ~MSTATUS_MPP_M) | MSTATUS_MPP_U;
  asm volatile(
      "csrw mstatus, %0;"
      "csrw mepc, %1;"
      "mret;"
      :
      : "r"(mstatus), "r"(test));
  return 0;
}
```

Compile the processor and the test program and run it with ibex and spike.
```
fusesoc --cores-root=. run --target=sim --setup --build lowrisc:ibex:ibex_simple_system --PMPEnable=1
make -C examples/sw/simple_system/hello_test
./build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/Vibex_simple_system -t --meminit=ram,./examples/sw/simple_system/hello_test/hello_test.elf ]
spike --isa=rv32imc --log-commits -d -m0x10000:0x30000,0x100000:0x100000 --priv=mu ./examples/sw/simple_system/hello_test/hello_test.elf
```

From the ibex and spike logs below, you can find the differences after the `mret` instruction.
* ibex log
```
174	        83	001003da	300027f3	csrrs	x15,mstatus,x0	  x0:0x00000000 x15=0x00000080
176	        84	001003de	7779	c.lui	x14,0xffffe	 x14=0xffffe000
178	        85	001003e0	7ff70713	addi	x14,x14,2047	 x14:0xffffe000 x14=0xffffe7ff
180	        86	001003e4	8ff9	c.and	x15,x14	 x15:0x00000080 x14:0xffffe7ff x15=0x00000080
182	        87	001003e6	00000717	auipc	x14,0x0	 x14=0x001003e6
184	        88	001003ea	f6670713	addi	x14,x14,-154	 x14:0x001003e6 x14=0x0010034c
186	        89	001003ee	30079073	csrrw	x0,mstatus,x15	 x15:0x00000080  x0=0x00000000
192	        92	001003f2	34171073	csrrw	x0,mepc,x14	 x14:0x0010034c  x0=0x00000000
194	        93	001003f6	30200073	mret	
208	       100	00100000	34e0006f	jal	x0,10034e	  x0=0x00000000
214	       103	0010034e	e9fff06f	jal	x0,1001ec	  x0=0x00000000
```

* spike log
```
core   0: >>>>  main
core   0: 0x00000000001003da (0x300027f3) csrr    a5, mstatus
core   0: 3 0x001003da (0x300027f3) x15 0x00000000
: 
core   0: 0x00000000001003de (0x00007779) c.lui   a4, 0xffffe
core   0: 3 0x001003de (0x7779) x14 0xffffe000
: 
core   0: 0x00000000001003e0 (0x7ff70713) addi    a4, a4, 2047
core   0: 3 0x001003e0 (0x7ff70713) x14 0xffffe7ff
: 
core   0: 0x00000000001003e4 (0x00008ff9) c.and   a5, a4
core   0: 3 0x001003e4 (0x8ff9) x15 0x00000000
: 
core   0: 0x00000000001003e6 (0x00000717) auipc   a4, 0x0
core   0: 3 0x001003e6 (0x00000717) x14 0x001003e6
: 
core   0: 0x00000000001003ea (0xf6670713) addi    a4, a4, -154
core   0: 3 0x001003ea (0xf6670713) x14 0x0010034c
: 
core   0: 0x00000000001003ee (0x30079073) csrw    mstatus, a5
core   0: 3 0x001003ee (0x30079073) c768_mstatus 0x00000000
: 
core   0: 0x00000000001003f2 (0x34171073) csrw    mepc, a4
core   0: 3 0x001003f2 (0x34171073) c833_mepc 0x0010034c
: 
core   0: 0x00000000001003f6 (0x30200073) mret
core   0: 3 0x001003f6 (0x30200073) c768_mstatus 0x00000080
: 
core   0: >>>>  test
core   0: 0x000000000010034c (0x00008082) ret
core   0: 0 0x0010034c (0x8082)
: 
core   0: 0x00000000001003c8 (0x000202b7) lui     t0, 0x20
core   0: 0 0x001003c8 (0x000202b7) x 5 0x00020000
: 
core   0: 0x00000000001003cc (0x000002a1) c.addi  t0, 8
core   0: 0 0x001003cc (0x02a1) x 5 0x00020008
```

GregAC, 2021-02-19:
Thanks for the detailed report @mmxsrup I'm not sure Spike is doing the right thing here, from the RISC-V privileged spec:

> If no PMP entry matches an M-mode access,  the access succeeds.  If no PMP entry matches an S-mode or U-mode access, but at least one PMP entry is implemented, the access fails

In effect if a core implements PMP and you wish to use U-mode you must use PMP. If your core doesn't implement PMP at all then U-mode still works.

Are you using the latest version of Spike? It could be worth filing an issue against the Spike repo and see if they agree with my interpretation of the spec.

mmxsrup, 2021-02-19:
@GregAC Thanks for the detailed explanation.

I'm using the latest version of spike.

From the description of the RISC-V privileged spec, I think the behavior of PMP in ibex is correct.
I'll submit this issue to Spike repo.

mmxsrup, 2021-02-22:
> Since the reset values for the PMP registers are platform-defined, both Spike and Ibex are doing something technically valid here.

It seems that the cause is the difference in the initial value depending on the platform.

closed 2021-02-22 by mmxsrup