lowRISC/ibex #1281: [Component:DV] Running regression riscv_bitmanip_full_test

mescoba1, 2021-02-23:
## My Environment
When running `make ITERATIONS=1 SIMULATOR=questa VERBOSE=1` the following test gets ran and errors out `riscv_bitmanip_full_test`, presumably since ibex implements RV32IMC and *not* RV32IMCB. Is this expected DV behavior?

**EDA tool and version:**
QuestaSim 2020.3

**Operating system:**
CentOS 7

**Version of the Ibex source code:**
N/A

mescoba1, 2021-02-23:
Bug in the [Makefile](https://github.com/lowRISC/ibex/blob/master/dv/uvm/core_ibex/Makefile#L43) perhaps?
```
# ISA
ISA                 := rv32imcb
```

rswarbrick, 2021-02-24:
This is a bit finicky. Ibex supports the bitmanip extension in some configurations, but we have to make sure the ISS runs in the same mode. My guess is that the two aren't in sync for you here. Can you share your log file?

mescoba1, 2021-02-24:
Which log file in particular?

GregAC, 2021-02-25:
The `regr.log` file you should find in the `out/seed-XXXX` directory should give the relevant error or failing that the log of the console output seen when running the test.

There are also per test logs in the `instr_gen/` and `rtl_sim/` directories. Generally it's the simulation log in `rtl_sim/[test]/sim.log` which contains things of interest.

rswarbrick, 2021-04-05:
Hi there! It's been quite a while with no response, so I'm going to close this issue now. If you come back to this and you're still having a problem, feel free to re-open (with the log file that Greg mentioned above).

closed 2021-04-05 by rswarbrick