lowRISC/ibex #1288: Hello there.

mike3s, 2021-02-26:
## My Environment

**EDA tool and version:**

**Operating system:**

**Version of the Ibex source code:**

mike3s, 2021-02-26:
I am trying to setup an ibex core using Vivado 2018.3 in windows 10.
No luck yet.
Is there any walkthrough or guide?

rswarbrick, 2021-02-27:
Hi there. Have you found our [user guide](https://ibex-core.readthedocs.io/en/latest/02_user/index.html)? You'll also probably want some form of simple SoC. We have an example for the Arty A7 development board: see the "examples" section of the user guide.

mike3s, 2021-02-27:
For now i dont want to use any SoC. I just want to simulate the core in Vivado. I also tried the example but I faced some errors. I will try again and I will post the exact procedure I follow to check if I done something wrong.
thank you,
Mike

rswarbrick, 2021-03-01:
Great, looking forward to hearing how things go!

closed 2021-03-01 by rswarbrick