lowRISC/ibex #1289: Purpose of current implementation of mip and mie

hoang26, 2021-02-26:
Hello, I'm taking a look at the the current ibex implementation for learning's sake and I'm curious if there's a reason why both mie and mip are type irqs_t. Is there a major benefit of this over defining it as 32-bit logic?

For example - if one wanted to perform any kind of logical operation of either of those registers against a 32-bit register, specific bits would need to be operated on instead of the entire register because it's non-standard.

## My Environment

**EDA tool and version:**

**Operating system:**

**Version of the Ibex source code:**
aa7582b8d420888541c2e8688447c17918a71aeb

rswarbrick, 2021-03-01:
Well, the code is going to want to set specific parts of the IRQ signals so it's nicer to have named fields than have to do calculations with indices. Also, these are packed structs, so you can treat them as just a bundle of wires if you want. For example, `ibex_cs_registers.sv` contains these lines:
```systemverilog
  assign irqs_o        = mip & mie_q;
  assign irq_pending_o = |irqs_o;
```
(where `irqs_o`, `mip` and `mie_q` have type `irqs_t`).

hoang26, 2021-03-02:
I can definitely see the readability advantage of using a packed struct instead of 32 logic and working with indices like the other CSRs. If using 32 bit logic for both mip and mie, the lines you mentioned seem like they'd also work with the reduction operator as well.

In addition to the irqs_t packed struct, mip is purely combinational here, while mie has both current and next state logic. Not sure if this warrants a different question thread, but if my understanding is correct, the primary reason why mip here doesn't have both a current and next state is because it's faster than waiting another clock cycle to exit WFI when an interrupt comes. While this works in the smaller ibex core, other implementations I see also have a current and next state mip register. Those implementations also have additional features, such as supervisor mode where mip is sequential logic.

rswarbrick, 2021-03-02:
@GregAC: You can probably answer this better than me?

GregAC, 2021-03-02:
`mip` is purely combinational so `irqs_o` can change state when Ibex is in sleep and the clock is disabled. `irqs_o` leads into the wake-up logic so were `irqs_o` based on a clocked `mip_q` we'd never wake-up. This also helps keep interrupt latency low.

It's possible this would cause timing issues in some systems but the IRQ lines could just be registered just outside the Ibex top level before they reach Ibex at the expense of one extra cycle of latency to fix these.

Ibex could register them internally, using seperate logic to observe the combinational state of the incoming interrupt for the wake-up logic or not disable the clock on the `mip_q` register however we don't have a strong reason to motivate this change.

It's possible some implementation wish to detect edges on their IRQ lines (Ibex interrupts purely on level based interrupts) where registering would be necessary. In more complex implementations you may also want finer control over when interrupts become architecturally visible (i.e. by reading the `mip` CSR) so you register `mip` to achieve this.

closed 2021-03-18 by GregAC