lowRISC/ibex #1294: Prim files generation for SW simulation

rahulraveendran15-coder, 2021-03-04:
HI,
I am planning to crate a standalone script for running the ibex SW  model without using fusesoc. Im seeing some files are getting generated using some python model. Right now I have copied the below files after running first time in fusesoc and then it ran fine. Rather than this approach I just want to create it alone without fusesoc core files. Could anyone please tell what can be the command for generating these files?
```

prim_clock_gating.sv  
prim_pkg.sv  
prim_ram_1p.sv  
prim_ram_2p.sv
```

rswarbrick, 2021-03-04:
Since fusesoc is part of the Ibex build system, we don't really support doing that.

The exact details of how fusesoc runs generators are a bit complicated. If you really, really want to auto-generate the code once, I suggest you run fusesoc and copy stuff out of the build directory. But it's probably a better idea to use the build system properly: are you doing something unusual? Maybe we can help.

rahulraveendran15-coder, 2021-03-04:
I was creating a standalone script using Xcelium for the sw model. Right now I build using fusesoc and copied the files and then ran my xcelium script which works fine. BUt since this prim files are generated as a part of build process I dont want it to be in my repo. If some one else wants to run my xcelium script each time they need to first build then copy/point to the prim files, whcih makes it a bit lengthy process.
My plan was to add the generation command  of the above files in my xcelium script so that we dont need to need to worry about  build process of fusesoc and the linking of files.(Getting rid of Fusesoc command)

rswarbrick, 2021-03-04:
The usual way to do this would be the other way around: teaching the overarching build system (which uses fusesoc) how to run Xcelium as a simulator. Indeed, that might even work already (edalize and dvsim definitely both have Xcelium backends). I think that I don't quite understand why the thing you're building has to be standalone, or indeed exactly what that means.

rahulraveendran15-coder, 2021-03-04:
In some of the systems we are restricted to use fusesoc and so creating the model in the existing methodology which we we are using.

rswarbrick, 2021-04-05:
I don't think that we're going to support anything other than using fusesoc for this. If you absolutely cannot run a small Python program for some reason, you'll have to do the hack you're describing.

closed 2021-04-05 by rswarbrick