lowRISC/ibex #1297: [epmp] question about handling mseccfg high bits

mundaym, 2021-03-04:
In the [ePMP specification](https://ibex-core.readthedocs.io/en/latest/_downloads/adbc55b5b9bad5f92968b58e7b12fc57/riscv-epmp.pdf) `mseccfg` is specified as a 64-bit register that is split into two parts (low at 0x390 and high at 0x391) on 32-bit architectures.

The Ibex documentation doesn't mention this so I am curious as to how/if we are currently handling the high bits. Should we document a `mseccfgh` zero read-only register at 0x391 so that portable software can read the full 64-bit value?

mundaym, 2021-03-04:
@GregAC

GregAC, 2021-03-04:
We're not handling the bits, that was my error. I had initially interpreted this as the high 32-bits were only implemented for RV64.

We should indeed have a `mseccfgh` read-only register at 0x391 that reads as all zeros. I'll do an Ibex issue to track.

PR #1310 (2021-03-19): [rtl] Add MSECCFGH CSR
This is the top 32 bits of MSECCFG. It currently has no specified bits
so reads as 0 and ignores writes.

Fixes #1297
closed 2021-03-19 by GregAC