lowRISC/ibex #1299: Yosys Synthesis Doubts

rahulraveendran15-coder, 2021-03-10:
Hi,
I have some doubts in Yosys synthesis.

(1) In synthesis description I saw the following line. Does it mean that apart from the .lib mentioned there I cannot use anthing else? Its written (**currently 'nangate' is the only supported value**). Basically if I want to use any other standardcell lib file what should I change in this parameter? Can I give it as my lib name which I am using? is this variable jst pointing to a name or is there anything more in it?

_LR_SYNTH_CELL_LIBRARY_NAME - The name of the standard cell library, this is used to alter the flow configuration for the library, currently 'nangate' is the only supported value_
Also for the same I can see in syn_setup.sh its asked to create a variable as below:
` export LR_SYNTH_CELL_LIBRARY_NAME=nangate`

(2) what is the use of prim_clock_gating.v and latch_map.v in synthesis?
```

yosys "read_verilog -sv ./rtl/prim_clock_gating.v $lr_synth_out_dir/generated/*.v"
# Map latch primitives onto latch cells
yosys "techmap -map rtl/latch_map.v"
```

GregAC, 2021-03-11:
1) I built the flow using the nangate library, there's no reason to believe it cannot work with other libraries but beyond a bit of work to switch out library names and paths I haven't put any work into support for other libraries. If you search through the tcl for 'cell_library_name' you will find the places it is used. Currently it's just for selecting an .sdc file. These contain references to specific cellset_driving_cell [all_inputs] -lib_cell BUF_X2s which need adjusting for different libraries, e.g. in `ibex_core.nangate.sdc` there is the following line:

```
set_driving_cell [all_inputs] -lib_cell BUF_X2
```
Switching to using a different .lib and producing .sdc files that reference the appropriate cells should be possible.

2) `prim_clock_gating.v` is a file required to build Ibex, it is used for the top-level clock gate the enables/disables the CPU clock. `latch_map.v` allow yosys to build the latch based synthesis file by telling it what latch cells to use. This too may require adjustment for library other than nangate.

rahulraveendran15-coder, 2021-03-17:
@GregAC 
Thank you very much .
Does the latch_map.v used because of refileLatch mode used in synthesis?

 I can see the synthesis is done using Yosys for regfileLatch. Is there any specific reason to use this configuration because in RTL  the default value of the corresponding parameter is regfileFF. But in synthesis we used regfileLatch. 

Also if we are using regfileFF configuration do we need to use latch_map.v?

GregAC, 2021-03-18:
> Does the latch_map.v used because of refileLatch mode used in synthesis?

Yes

> I can see the synthesis is done using Yosys for regfileLatch. Is there any specific reason to use this configuration because in RTL the default value of the corresponding parameter is regfileFF. But in synthesis we used regfileLatch.

The latch based register file consumes less area

> Also if we are using regfileFF configuration do we need to use latch_map.v?

I don't think so

tomroberts-lowrisc, 2021-03-18:
There is also a latch in the clock gating cell which will need to be mapped somehow.

rswarbrick, 2021-04-05:
Hi! I'm closing this because I think the question has been addressed. If there's something missing, please re-open.

closed 2021-04-05 by rswarbrick