lowRISC/ibex #1302: Riscv_Arithmetic_Basic_TEST Failure

ishita71, 2021-03-11:
Hi,

I tried to run the arithmetic basic test but the test got failed.
[image]

I have attached regr.log and spike_sim.log
[riscv_arithmetic_basic_test.0.log](https://github.com/lowRISC/ibex/files/6123311/riscv_arithmetic_basic_test.0.log)

[regr.log](https://github.com/lowRISC/ibex/files/6123307/regr.log)

Can you help me to solve the above issue?

Regards,
Ishita

GregAC, 2021-03-11:
Are you using the lowRISC branch of spike: https://github.com/lowRISC/riscv-isa-sim/tree/ibex?

ishita71, 2021-03-23:
Hi GregAC,

Sry for the late reply

Yes I am using lowRISCV branch of spike

GregAC, 2021-03-26:
The log you've provided is just the spike log so can't determine what might be going wrong from that alone. This is a trace mismatch so you need to look at both the spike output and the RTL simulator output (within a seed-XXXX directory spike outputs are found in instr_gen/spike_sim and the RTL outputs are found in rtl_sim/[test_name]/).

Though the arithmetic basic test doesn't have any known issues. You've only got one iteration here does it always fail if you have more iterations or is it a rare failure?

Are you using the latest version of Ibex? Have you made any modifications? Which simulator and what version of that simulator are you using?

ishita71, 2021-03-31:
I have run this test in master branch commit: 69ae65c71352028cd175fe3ce99f2404f46b14c5 
I am using a spike simulator and for the version, I am not sure how to check it.

I have attached log files please let me know if any futher details needed
[compile.log](https://github.com/lowRISC/ibex/files/6236529/compile.log)
[regr.log](https://github.com/lowRISC/ibex/files/6236530/regr.log)
[sim_riscv_arithmetic_basic_test_0.log](https://github.com/lowRISC/ibex/files/6236531/sim_riscv_arithmetic_basic_test_0.log)
[sim.log](https://github.com/lowRISC/ibex/files/6236533/sim.log)
[trace_core_00000000.log](https://github.com/lowRISC/ibex/files/6236534/trace_core_00000000.log)

rswarbrick, 2021-04-05:
Hi there! This test passes for me. You didn't include the trace from the Spike run (it's at `out/seed-1767/instr_gen/spike_sim/riscv_arithmetic_basic_test.0.log`) but something looks very odd in `regr.log` - it appears that the first divergence between the traces has Spike at a different address from Ibex (`0x80001780` instead of `0x80000276`). The previous instruction is a straight-line `addi` instruction, so my guess is that your Spike has taken some sort of exception.

It's kind of hard to debug further without more information: could you include the Spike log I mentioned above? Also, it might be worth making really sure you're using the version of Spike that you think you're using.

closed 2021-12-10 by GregAC