lowRISC/ibex #1307: Usage of default riscv-isa-sim spike possible?

cr8601, 2021-03-16:
Hi,
I tried to run DV with the standard spike (riscv-isa-sim). Unfortunately it fails and I can see that the customs CSR are obviously the source root of the problem since they do not exist in github.com/riscv/riscv-isa-sim.

Since I am currently also dealing with other CPUs than ibex, my intital plan was to use the "master" spike version from github.com/riscv/riscv-isa-sim to simulate them all

- This seems to be impossible. Correct?

For my understanding it would be kind, if some could answer the following questions where I think they are easy for some who is more involved into these topics:

- Does each CPU implementation with custom CSRs do need its own spike branch? (this is probably valid for every CPU?)

- In that case the github.com/riscv/riscv-isa-sim is only the master branch for all cpu specific branches and not to be used for non-CPU-developers?

- What about other simulators like ovpsim? Is there any ibex support?

- Can I 'make' the lowrisc/riscv-isa-sim 'standalone'? Typically I do '../configure --prefix=$RISCV' where RISCV points to my 'general' toolchain. There I do not want to include an CPU specific spike...

Thanks and best regards
cr8601

GregAC, 2021-03-16:
> This seems to be impossible. Correct?

It is possible, provided you don't want to test the ICache. I believe it will work with a single change just remove the function `init_custom_csr` from `dv/uvm/core_ibex/riscv_dv_extension/ibex_asm_program_gen.sv` entirely

This is the function:
https://github.com/lowRISC/ibex/blob/62b1a30c7d0d8fa60a0476d896fec7551cb4c96f/dv/uvm/core_ibex/riscv_dv_extension/ibex_asm_program_gen.sv#L37-L40

Without that the ICache CSR write won't be emitted and it should work with plain spike, though beware if you're also using bitmanip you may find some mismatches as the spec is still evolving.

I have been meaning to do some work so there will be a config option to do this for you. In the mean time it'd be worthwhile for us to document this fix.

> Does each CPU implementation with custom CSRs do need its own spike branch? (this is probably valid for every CPU?)

I'm not familiar enough with Spike to comment but my understanding was it doesn't have any out of the box support for custom CSRs. So any CPU that uses Spike as a model and wants to stimulate custom CSRs would require a custom Spike.

> In that case the github.com/riscv/riscv-isa-sim is only the master branch for all cpu specific branches and not to be used for non-CPU-developers?

I'm not quite sure what you're getting at with this question. The master branch of github.com/riscv/riscv-isa-sim will not be CPU specific as it's the main spike repository.

> What about other simulators like ovpsim? Is there any ibex support?

I did think OVPSim could deal with the custom CSR but not sure off-hand @udinator will know more

> Can I 'make' the lowrisc/riscv-isa-sim 'standalone'? Typically I do '../configure --prefix=$RISCV' where RISCV points to my 'general' toolchain. There I do not want to include an CPU specific spike...

If you change the prefix you can place the Ibex Spike elsewhere e.g. do `../configure --prefix=$RISCV/ibex` just remember to set the `SPIKE_PATH` environment variable appropriately when running Ibex DV.

cr8601, 2021-03-17:
Thanks for that comments, it helps a lot for a better understanding!

closed 2021-03-17 by cr8601

udinator, 2021-03-17:
Just to confirm, @GregAC is correct that OVPsim can currently not handle the custom CSRs implemented in the core - so if you want to use that ISS please make sure to disable the `init_custom_csr` function and use the correct core config.