lowRISC/ibex #1308: simple system Example

xvirus313, 2021-03-16:
hello.
I am trying to simulate the simple system example.
I downloaded https://github.com/lowRISC/lowrisc-toolchains/releases and I extracted it in desktop. 
When I do 
"make -C examples/sw/simple_system/hello_test" 
I take this error:
  
make -C examples/sw/simple_system/hello_test
make: Entering directory '/home/x-virus/Desktop/ibex-master/examples/sw/simple_system/hello_test'
riscv32-unknown-elf-gcc -march=rv32imc -mabi=ilp32 -static -mcmodel=medany -Wall -g -Os -fvisibility=hidden -nostdlib -nostartfiles -ffreestanding  -MMD -c -I/home/x-virus/Desktop/ibex-master/examples/sw/simple_system/common -o /home/x-virus/Desktop/ibex-master/examples/sw/simple_system/common/simple_system_common.o /home/x-virus/Desktop/ibex-master/examples/sw/simple_system/common/simple_system_common.c
make: riscv32-unknown-elf-gcc: Command not found
make: *** [/home/x-virus/Desktop/ibex-master/examples/sw/simple_system/hello_test/../common/common.mk:65: /home/x-virus/Desktop/ibex-master/examples/sw/simple_system/common/simple_system_common.o] Error 127

the "riscv32-unknown-elf-gcc" that says Command not found, is in the bin folder of toolchain. 

I tryed to copy bin folder in ibex-master/build/lowrisc_ibex_ibex_simple_system_0 or change the path in  common.mk 

nothing works.
Any help please?

rswarbrick, 2021-03-17:
You'll need to make sure that your toolchain bin directory is on your $PATH. This is a general unix thing, so you can find proper information about it elsewhere. Basically, if you're using the bash shell you want to run something like `export PATH=/path/to/toolchain/bin:$PATH`. Then run `which riscv32-unknown-elf-gcc` to check the change worked. If it prints the path to the toolchain, you should be good to go. You'll also want to add that line to `.bashrc` so that it works in new shells too.

xvirus313, 2021-03-18:
I fixed the path but still when I try to make -C examples/sw/simple_system/hello_test
It goes:

make: Entering directory '/home/x-virus/Desktop/ibex-master/examples/sw/simple_system/hello_test'
riscv32-unknown-elf-gcc -march=rv32imc -mabi=ilp32 -static -mcmodel=medany -Wall -g -Os -fvisibility=hidden -nostdlib -nostartfiles -ffreestanding -MMD -c -I/home/x-virus/Desktop/ibex-master/examples/sw/simple_system/common -o /home/x-virus/Desktop/ibex-master/examples/sw/simple_system/common/simple_system_common.o /home/x-virus/Desktop/ibex-master/examples/sw/simple_system/common/simple_system_common.c
****make: riscv32-unknown-elf-gcc: Command not found****
make: *** [/home/x-virus/Desktop/ibex-master/examples/sw/simple_system/hello_test/../common/common.mk:65: /home/x-virus/Desktop/ibex-master/examples/sw/simple_system/common/simple_system_common.o] Error 127

rswarbrick, 2021-03-18:
What happens when you run `which riscv32-unknown-elf-gcc`? This is not an Ibex-specific problem: it's a problem that you're facing when installing a package onto your system and updating your path to find it.

xvirus313, 2021-03-18:
nothing happens.
Do I have to run cmake for riscv32-unknown-elf-gcc.cmake?

rswarbrick, 2021-03-18:
If nothing happens, that means you haven't updated your path correctly. You'll need to fix this. I'm going to close this issue now, because the problem is that you are struggling with getting your environment right and this isn't anything to do with Ibex in particular. Please re-open, or open another issue, if you find another problem.

closed 2021-03-18 by rswarbrick