lowRISC/ibex #1311: [dv] Spike timeout in riscv_mmu_stress_test in experimental-maxperf-pmp-bmfull-icache

imphil, 2021-03-22:
* Configuration: experimental-maxperf-pmp-bmfull-icache
* Test: riscv_mmu_stress_test
* stdout
	  ```
	  2021-03-21T20:31:25.8153074Z Sun, 21 Mar 2021 20:31:25 ERROR    Timeout[10s]: /nas/lowrisc/tools/spike-ibex/20201023-git-255bf1cacc599b1413438c269100f3ecd0eb3352/bin/spike --log-commits --isa=rv32imcb -l out/seed-27831/instr_gen/asm_test/riscv_mmu_stress_test_9.o &> out/seed-27831/instr_gen/spike_sim/riscv_mmu_stress_test.9.log
	  ```

* `regr.log`:
  
	  ```
	  riscv_mmu_stress_test.9
	  -----------------------
	  Test binary: out/seed-27831/instr_gen/asm_test/riscv_mmu_stress_test.9.o
	  sim log: out/seed-27831/rtl_sim/riscv_mmu_stress_test.9/sim.log
	  ibex : out/seed-27831/rtl_sim/riscv_mmu_stress_test.9/trace_core_00000000.csv
	  spike : out/seed-27831/instr_gen/spike_sim/riscv_mmu_stress_test.9.csv
	  Mismatch[1]:
	  ibex[34685] : pc[8003107e] lb   x20,883(x10): s4:00000000
	  spike[34685] : pc[ffffffff80001680] addi    s8, s8, -124: s8:80047d6c
	  Mismatch[2]:
	  ibex[34686] : pc[80031082] c.srai   x14,0x11: a4:00000000
	  spike[34686] : pc[ffffffff80001700] csrr    a3, mepc: a3:8003106a
	  Mismatch[3]:
	  ibex[34687] : pc[80031084] c.mv x9,x29: s1:8004bce4
	  spike[34687] : pc[ffffffff80001704] csrr    a3, mcause: a3:00000007
	  Mismatch[4]:
	  ibex[34688] : pc[80031086] lw   x7,848(x10): t2:00000000
	  spike[34688] : pc[ffffffff80001708] c.li    s10, 3: s10:00000003
	  Mismatch[5]:
	  ibex[34689] : pc[8003108e] slti x27,x30,-112: s11:00000000
	  spike[34689] : pc[ffffffff80001712] c.li    s10, 8: s10:00000008
	  242436 instructions left in trace spike
	  [FAILED]: 22982 matched, 245468 mismatch
   
* CI Run: https://dev.azure.com/lowrisc/lowrisc-private/_build/results?buildId=36778&view=logs&j=dace1ca7-cf0b-575c-3ae3-5c7773998304
* All log files: [timeout-riscv_mmu_stress_test.zip](https://github.com/lowRISC/ibex/files/6183564/timeout-riscv_mmu_stress_test.zip)

@udinator would you mind taking a look?

imphil, 2021-03-22:
Might be the same thing as https://github.com/lowRISC/ibex/issues/1165

udinator, 2021-03-25:
hmm, will take a look at this.

GregAC, 2021-12-10:
Closing without resolution as upcoming changes with the testbench and co-simulation system will alter things. We'll do a new pass over regression failures once that work is complete.

closed 2021-12-10 by GregAC