lowRISC/ibex #1313: [dv] riscv_debug_ebreakmu_test: Core did not enter debug mode after execution of ebreak

imphil, 2021-03-22:
* Config: experimental-maxperf-pmp-bmfull-icache
* `regr.log`: 
```
riscv_debug_ebreakmu_test.9
---------------------------
Test binary: out/seed-27831/instr_gen/asm_test/riscv_debug_ebreakmu_test.9.o
sim log: out/seed-27831/rtl_sim/riscv_debug_ebreakmu_test.9/sim.log
UVM_FATAL /azp/agent/_work/1/s/dv/uvm/core_ibex/tests/core_ibex_base_test.sv(204) @ 8886400: reporter [uvm_test_top] Check failed signature_data == core_status (8 [0x8] vs 1 [0x1]) Core did not enter debug mode after execution of ebreak
[FAILED]: sim error seen
```
* Log files and waves: [simerror-riscv_debug_ebreakmu_test.zip](https://github.com/lowRISC/ibex/files/6183819/simerror-riscv_debug_ebreakmu_test.zip)

@udinator @GregAC

GregAC, 2021-03-22:
Was there just this single failure in the run or did all of them fail? Regression run I just completed using #1309 passed all iterations of this test.

I'll do a bunch of iterations of this test specifically with latest master to see if it's still occurring.

imphil, 2021-03-22:
The `riscv_debug_ebreakmu_test` test ran 15 times, one out of those 15 failed, all other 14 were successful.

GregAC, 2021-03-24:
I had a go at recreating this with a local run and found 1 failure in 30 that I thought was the same, however just seen they're subtly different but may have the same root cause. First the issue I saw:

It turned out to be an interesting corner case and looks like Ibex is doing the correct thing and the testbench is getting confused.

The test first stimulates a `debug_req` to enter debug mode so it can set the bit in `dcsr` that turns `ebreak` into a breakpoint (so it enters debug mode rather than just causing an exception much like an `ecall`).

Occasionally that initial `debug_req` will occur the same cycle the first `ebreak` is executed. Ibex correctly sets up the `mstatus`, `mcause`, `mepc` for the exception the `ebreak` should cause then immediately enters debug mode setting up the debug registers based upon the exception entry (so return PC and privilege levels are the ones required for the exception handler entry). The testbench isn't expecting this and in particular expects a normal debug mode entry for the `debug_req` from U-mode in the failing case so complains `dcsr.prv` has the wrong value.

The failure listed here isn't complaining about `dcsr.prv` but rather a failure to enter debug mode after `ebreak`. It may well be a similar issue with slightly different timing, I'll take a look at the waves to confirm.

Fix wise it may be simplest to ensure the `debug_req` happens before any `ebreak` is executed. Having them occur together is a good test but not one this was intended to simulate and adapting it to deal with the scenario may be tricky. The test library refactoring I'm considering should make testing these kinds of scenarios simpler.

GregAC, 2021-03-24:
Yes waves from initially presented issue show a different problem but same root cause. This time the `ebreak` executes and begins fetching the exception handler just before `debug_req` is raised. Debug entry proceeds as normal which the testbench is happy with, but after the `dret` we continue running the exception handler. The handler outputs some signature data out which the testbench doesn't consume as it wasn't expecting exception handler entry. It then waits for another `ebreak` that sees and then immediately consumes some signature data and then fails as the signature data relates to exception handler entry not `ebreak` driven debug mode entry the testbench was expecting.

Again adapting the checking to allow this will be tricky and ensuring `debug_req` happens to set things up before executing `ebreak` is a simpler fix whilst we consider more flexible checking techniques.

GregAC, 2021-03-25:
After discussion as this failure is reasonably rare and even the quicker fix (stop the tricky scenario being stimulated) may involve reasonable work we'll leave it unfixed for now pending wider refactoring work for the tests.

udinator, 2021-03-25:
@GregAC I took a look at the waves as well and came to the same root cause as you did.
I agree that this is a tricky edge case scenario to support in the testbench, so might be worth including it in any wider refactoring work.

imphil, 2021-03-25:
It would be great if we can somehow either manually or automatically identify this special case quickly: if we run the regression frequently (as we hope to do) it's easy to judge if it is pass or fail; it is much harder, however, to look at each failure and then decide it's a well-known failure.

Just ignoring all failures of riscv_debug_ebreakmu_test seems overly broad?

GregAC, 2021-03-26:
We could add functionality to sim.py so it can take a list of known test failure signatures that is ignores for the purposes of overall pass/fail. Something like a regex and a test name and if the failure reason matches the test name the failure is ignored.

The issue here is the message we'd ignore could either be related to this bug or another failure, can't really tell. I can maybe quickly add something to the test that sees we've had the `debug_req` close to `ebreak` issue and just fails early with a specific message. You could then filter on that.

imphil, 2021-03-26:
> I can maybe quickly add something to the test that sees we've had the debug_req close to ebreak issue and just fails early with a specific message. You could then filter on that.

Something like that would be great!

GregAC, 2021-12-10:
Upcoming testbench changes with the new co-simulation system will fix this.

closed 2021-12-10 by GregAC