lowRISC/ibex #1314: VCS reported an error, NYI-CSTR-SYS-FTC

SerLippo, 2021-03-23:
I am totally a rookie and learning this.
I did nothing but download the toolchains and build the spike under the user guide.
When I type "make" in the terminal.
My VCS just report an error like this.

## My Environment

**EDA tool and version:**

**Operating system:**

**Version of the Ibex source code:**

rswarbrick, 2021-03-23:
I suspect we just require a newer version of VCS than that. I see that we don't list a minimum version in our tool requirements page (sorry!), but we've been using the "Q-2020.03" series of releases to test things locally. I'll file a follow-up issue to track adding a version to the requirements page.

SerLippo, 2021-03-23:
> I suspect we just require a newer version of VCS than that. I see that we don't list a minimum version in our tool requirements page (sorry!), but we've been using the "Q-2020.03" series of releases [...]

thanks, I will updata my VCS and try again.

closed 2021-04-06 by rswarbrick