lowRISC/ibex #1317: [Component:DV] Possible False Positive Pass Results for Tests

mescoba1, 2021-03-25:
when I run the following test:
```
make SEED=123 TEST=riscv_debug_basic_test
```

the regression log shows zero match instructions:
```
$ cat out/seed-123/regr.log 

riscv_debug_basic_test.0
------------------------
Test binary: out/seed-123/instr_gen/asm_test/riscv_debug_basic_test.0.o
sim log: out/seed-123/rtl_sim/riscv_debug_basic_test.0/sim.log
ibex : out/seed-123/rtl_sim/riscv_debug_basic_test.0/trace_core_00000000.csv
spike : out/seed-123/instr_gen/spike_sim/riscv_debug_basic_test.0.csv
[PASSED]: 0 matched
1 PASSED, 0 FAILED
```
the reason for
 no matched instructions is due to the `spike_log_to_trace_csv.py` script not dumping the instructions to the 2nd column.
```
$ head out/seed-123/instr_gen/spike_sim/riscv_debug_basic_test.0.csv
pc,instr,gpr,csr,binary,mode,instr_str,operand,pad
ffffffff80000080,,sp:40001000,,40001137,3,"lui     sp, 0x40001",,
ffffffff80000084,,sp:40001106,,10610113,3,"addi    sp, sp, 262",,
ffffffff8000008c,,s10:8002308c,,00023d17,3,"auipc   s10, 0x23",,
ffffffff80000090,,s10:80022ce4,,c58d0d13,3,"addi    s10, s10, -936",,
ffffffff80000094,,sp:80002094,,00002117,3,"auipc   sp, 0x2",,
ffffffff80000098,,sp:80002400,,36c10113,3,"addi    sp, sp, 876",,
ffffffff8000009c,,sp:80002401,,00116113,3,"ori     sp, sp, 1",,
ffffffff800000a4,,s5:800030a4,,00003a97,3,"auipc   s5, 0x3",,
ffffffff800000a8,,s5:80003134,,090a8a93,3,"addi    s5, s5, 144",,
```

Is this known behavior? Could this lead to false positive regression results? The two csv files look to be equivalent, and no failures are flagged in the riscv-dv compare script... 

I noticed the same behavior for the following tests as well.
```
riscv_debug_basic_test.0
riscv_debug_branch_jump_test.0
riscv_debug_instr_test.0
riscv_debug_wfi_test.0
riscv_debug_ebreak_test.0
riscv_debug_csr_entry_test.0
riscv_irq_in_debug_mode_test.0
riscv_debug_in_irq_test.0
riscv_single_interrupt_test.0
riscv_multiple_interrupt_test.0
riscv_nested_interrupt_test.0
riscv_interrupt_instr_test.0
riscv_interrupt_wfi_test.0
riscv_interrupt_csr_test.0
```

GregAC, 2021-03-25:
Various tests disable the cycle by cycle ISS and RTL log comparisons and just compare final state instead. If you look in dv/uvm/core_ibex/riscv_dv_extension/testlist.yaml you can see some have `compare_final_value_only: 1` set. 

These occur in tests around interrupts and debug behavior as you can't stimulate spike to produce identical traces.

It's should be possible to modify this so trace comparison can occur outside of interrupt/debug handlers at least. Something we'd like to take a look at in the future.

mescoba1, 2021-04-01:
that makes sense, thanks for the clarification

closed 2021-04-01 by mescoba1