lowRISC/ibex #1328: clock for simple system

rahulraveendran15-coder, 2021-04-08:
Hi,

I just want to know the clock period used for ibex simple system.  From which file the IO_CLK inside ibex_simple_system.sv is getting value when I am using Verilator.

I can see the below point where the top level binding is done:

Where can I see the toggling or period setting of IO_CLK
In the below sections there is nothing regarding clock values.

```
int main(int argc, char **argv) {
  ibex_simple_system top;
  VerilatorMemUtil memutil;
  VerilatorSimCtrl &simctrl = VerilatorSimCtrl::GetInstance();
  simctrl.SetTop(&top, &top.IO_CLK, &top.IO_RST_N,
                 VerilatorSimCtrlFlags::ResetPolarityNegative);
```

```
void VerilatorSimCtrl::SetTop(VerilatedToplevel *top, CData *sig_clk,
                              CData *sig_rst, VerilatorSimCtrlFlags flags) {
  top_ = top;
  sig_clk_ = sig_clk;
  sig_rst_ = sig_rst;
  flags_ = flags;
}
```

rswarbrick, 2021-04-08:
One option would be to run the simulation and look at the waveform. Another would be to note that the clock period doesn't matter: this design has a single clock, so it could be 1GHz or 1KHz and the RTL would simulate exactly the same way.

rahulraveendran15-coder, 2021-04-08:
So if  want to change the period how can I achieve it  when using verilator? In that case which parameter I should change?
I just want to calculate the execution time from the cycles number.

rahulraveendran15-coder, 2021-04-08:
I found from waveform the clock period is 2ps, but how and where it is setting can I get some idea?

GregAC, 2021-04-08:
The 2ps comes from Verilator. It appears to consider each time we call the top-level `eval()` 1ps of simulation time. To produce the clock we toggle IO_CLK every time we call `eval()` so you get one positive clock edge every other `eval()` call giving you a 2ps clock period.

It may be possible to alter the default simulation timeslice with the `--timescale-override` option in Verilator (see the documentation here: https://www.veripool.org/projects/verilator/wiki/Manual-verilator), for simple system this would need to be added to the file examples/simple_system/ibex_simple_system.core

However as @rswarbrick says the period is just an arbitrary number, the important thing is the number of clock cycles something takes to execute which will remain the same regardless of what timescale Verilator uses.

To get a wall-clock execution time you need to choose a frequency a real implementation of Ibex might run at. For an FPGA Implementation you can run the FPGA Implementation flow and see what maximum frequency the tools will give you (probably 100 MHz or less). For ASIC it will go faster (maybe several hundred MHz depending upon the silicon library used).

vogelpi, 2021-05-19:
The original questions seems to have been answered and there hasn't been further discussion here for more than a month. I am thus closing the issue. Please feel free to reopen it.

closed 2021-05-19 by vogelpi