lowRISC/ibex #1330: Porting registers from IMC TO EMC

rahulraveendran15-coder, 2021-04-12:
Hi,

I am trying to run some software models on RV32EMC using IBEX. I got the following errors from the crt0.S file.
How can I replace the below registers which is above x15 to make it compatible with EMC?As you know in EMC registers x16 to x31 are not available.

_start:
  .global _start

  /* clear BSS */
  la x26, _bss_start
  la x27, _bss_end

  bge x26, x27, zero_loop_end

zero_loop:
  sw x0, 0(x26)
  addi x26, x26, 4
  ble x26, x27, zero_loop
zero_loop_end:

rswarbrick, 2021-04-12:
Well, you'll have to replace `x26` and `x27` with registers below `x15`. I don't think that we're using any other registers at this point, so e.g. `x14` and `x15` should work just fine. If they don't, I think you'll have to do some debugging: the EMC configuration isn't something we are putting effort into supporting at the moment.

vogelpi, 2021-05-19:
The original questions seems to have been answered and there hasn't been further discussion here for more than a month. I am thus closing the issue. Please feel free to reopen it.

closed 2021-05-19 by vogelpi