lowRISC/ibex #1335: Usage of prim_clock_gating and latch_map.v during synthesis

rahulraveendran15-coder, 2021-04-13:
(1) Do I need to use prim_clock_gating.v and latch_map.v file during synthesis when using RegfileFF as my configuration?
Also In reference with closed case #1299 @GregAC  told that 

`prim_clock_gating.v is a file required to build Ibex, it is used for the top-level clock gate the enables/disables the CPU clock. latch_map.v allow yosys to build the latch based synthesis file by telling it what latch cells to use. This too may require adjustment for library other than nangate.`

If answer for (1) is no, then if I want to use RegfileLatch do I need to change both files if I am using a differenct library for synthesis?changing latch_map.v makes sense. What about prim_clock_gating.v, I can't see any latch maping there?

vogelpi, 2021-04-14:
Hi @rahulraveendran15-coder,

I am pretty sure that you need both `prim_clock_gating.v` and `latch_map.v` independent of the selected register file. As you said `latch_map.v` is needed to tell Yosys which cell to map generic latches to that are inferred when synthesizing the Verilog code. The `prim_clock_gating.v` is just a latch-based clock gate. Without `latch_map.v` Yosys most likely won't know to which cell of the library it should map the latch inside `prim_clock_gating.v` during the technology mapping stage.

Now, if you want to switch to a different library, you just need to change `latch_map.v`.

rahulraveendran15-coder, 2021-04-14:
Hi @vogelpi 
So are you telling eventhough if I am using the RegfileFF , do  I need to map the latch cell inside my library to the latch_map.v for prim_clock_gating.v file proper working(since clock gating is needed in all cases.)

Also how it is mapping the latches, I can see the module name for latch_map.v as like below. what will be the module name here? Where this module is instantiated?

module $_DLATCH_P_ (input E, input D, output Q);

rahulraveendran15-coder, 2021-04-14:
Also Greg told that "latch_map.v used because of refileLatch mode used in synthesis". In that case does it needed for RegfileFF?
#1299

vogelpi, 2021-04-14:
> So are you telling eventhough if I am using the RegfileFF , do I need to map the latch cell inside my library to the latch_map.v for prim_clock_gating.v file proper working(since clock gating is needed in all cases.)

Correct, even though you are using RegFileFF, you'll need `latch_map.v` for `prim_clock_gating.v`.

> Also how it is mapping the latches, I can see the module name for latch_map.v as like below. what will be the module name here? Where this module is instantiated?
>
> module $_DLATCH_P_ (input E, input D, output Q);

During synthesis, usually steps like this happen:
1. The synthesis tool (Yosys) firsts read the Verilog source code which can still be behavioral. For example `prim_clock_gating.v` is behavioral code. The code describes the behavior of a latch but it doesn't instantiate a latch cell (this would be structural code).
2. The synthesis tool translates the code and maps it to an internal representation on which it can perform some optimization steps etc.
3. At some point this internal representation is mapped to a structural representation, meaning a netlist that instantiates and connects cells. These cells are still generic: they don't belong to specific technology library yet but they encapsulate the functionality of gates. One example for such a generic cell is the `$DLATCH_P` cell you've mentioned. Basically, this generic netlist can be viewed as a human readable format of the internal representation.
4. Ultimately, the generic netlist is mapped to the target technology library. This means for every generic cell in the netlist, the synthesis tool will have to decide which cell from the technology library to use. And that's where `latch_map.v` comes in. Yosys has the `$DLATCH_P` cell in the generic netlist and when mapping it to the nangate45 library, will insert a `DLH_X1` cell.

GregAC, 2021-04-14:
Note that if you don't want the latch you can always modify `prim_clock_gating.v` so it ignore the enable signal and just passes the clock straight through. Of course this means you won't have a top-level clock gate for Ibex any more whether this matters depends on what you're doing.

rahulraveendran15-coder, 2021-04-14:
@vogelpi  How to see the generic cell "DLATCH_P"in the premapped netlist. I am not seeing

ALso in generic or premapped netlist and post synthesis netlist I am seeing "DLH_X1" instance. As you said it should't come in premapped netlist

tomroberts-lowrisc, 2021-04-14:
The latch mapping is happening before we write out the pre-mapped netlist in our synthesis script. Not sure why I did that! but I'll fix it now.

rahulraveendran15-coder, 2021-04-14:
I tried running the synthesis with the latest fix, still DMATCH_P is not there in my premapped netlist, then how the mapping is done with latch instance inside library file?

vogelpi, 2021-04-14:
Could you please attach the premapped netlist here?

rahulraveendran15-coder, 2021-04-14:
[ibex_core.pre_map.v.zip](https://github.com/lowRISC/ibex/files/6310491/ibex_core.pre_map.v.zip)
Please find the attachment

tomroberts-lowrisc, 2021-04-14:
I think you might be using an old version of Ibex where a dummy version of prim_clock_gating.v was used for synthesis (which just bypassed the clock gate). On the current master, I now see $DLATCH_P in the premapped netlist

rahulraveendran15-coder, 2021-04-14:
[ibex_top.pre_map.v.zip](https://github.com/lowRISC/ibex/files/6310661/ibex_top.pre_map.v.zip)
I tried running in new database. Still I am not able to find $DLATCH_P in the premapped netlist.(I am grepping for the word) PLease correct me if I am wrong. Premapped netlist is attached.
Also the prim_clock_gating.v file in my olda dataabase and new database is exactly same

vogelpi, 2021-04-14:
Thanks for attaching the pre-mapped netlist, that's useful. I think the clock gate is there, but it's somehow not there as a separate latch cell. For example, on line 32910:
```verilog
  assign \core_clock_gate_i.clk_o  = clk_i & \core_clock_gate_i.en_latch ;
```
And further on line 40431:
```verilog
  (* src = "syn_out/ibex_14_04_2021_14_01_27/generated/ibex_top.v:148.20-153.3|./rtl/prim_clock_gating.v:16.3-20.6" *)
  always @*
    if (!clk_i) \core_clock_gate_i.en_latch  = _00027_;
```
So the latch is there but its still behavioral code. I am not sure if Yosys will be able to map this to the DLH_X1 cell using `latch_map.v` and what it will do with that instead.

tomroberts-lowrisc, 2021-04-14:
Hmmm, what version of Yosys are you using?
I'm using:
```
$ yosys -version
Yosys 0.9+2406 (git sha1 3d8d98d7, gcc 8.4.0-1ubuntu1~18.04 -fPIC -Os)
```
If you're using a newer version they might have updated how latches are dealt with maybe?

vogelpi, 2021-05-19:
The original questions seems to have been answered and there hasn't been further discussion here for more than a month. I am thus closing the issue. Please feel free to reopen it.

closed 2021-05-19 by vogelpi