lowRISC/ibex #1340: Debug support for ibex

mrdiogodias, 2021-04-15:
Hello,
I want to use the ibex core, however, I need a debugger. For that reason, I'm using the code from #1218. 
I managed to port the code to my fpga but now I'm a little lost on how can I debug the led application example.
I already understood that I need an Olimex adapter but besides that what do I need to do in order to use gdb? 
Sorry for my ignorance but I'm new to the field.

Thank you

GregAC, 2021-04-17:
Hi @mrdiogodias I'm afraid we can't offer much in the way of help as we keep this repository focused on Ibex and how you build/use Ibex systems is a different topic. In general you'll need some kind of debug probe (JTAG is commonly used and that's what's used in that PR) to connect to the board then something to bridge GDB and the debug probe. OpenOCD is commonly used for this (http://openocd.org/). It connects to the debug probe and provides a network port that GDB can connect to.

You could take a look at OpenTitan (https://github.com/lowRISC/opentitan) which uses Ibex and contains debug functionality (https://docs.opentitan.org/doc/ug/getting_started_fpga/ shows how you connect to it under 'Connect with OpenOCD and debug' ) however it's a complex SoC so not a good minimal example.

Also consider if you really need the debugger, it's useful to have but you can do a lot with FPGA logic debugging instead (e.g. if using Xilinx and Vivado see https://www.xilinx.com/video/hardware/logic-debug-in-vivado.html) or just use printf debugging via a UART.

mrdiogodias, 2021-04-19:
Thank you for your explanation, I will take a look into those topics!

rswarbrick, 2021-05-21:
Hi there! I'm closing this issue, since the question seems to be answered. Please feel free to get in touch again with anything else.

closed 2021-05-21 by rswarbrick