lowRISC/ibex #1342: [dv] Enable ISS compare for at least some debug/interrupt/exception tests

GregAC, 2021-04-16:
Currently we disable instruction by instruction ISS/RTL comparison when interrupts, debug requests and some tests involving exceptions. This is because the jump to the exception handler will only occur in RTL and not the ISS as it's due to some external event.

Whilst final state comparison does occur this means we can miss some issues. For instance the bug described here: https://github.com/lowRISC/ibex/issues/1339 is fixed in current Ibex master but after some investigation where I intentionally recreated the problem a run of riscv_single_interrupt_test sees the bug but the test passes anyway. A load is repeated twice but as it's using different source and destination registers no register corruption occurs, so the second one just loads the same data from the same address.

Two load instructions are seen in the Ibex log where there should be just one. Had their been an ISS <-> RTL log comparison the bug would have been spotted and the test failed.

Given sufficient test iterations it is likely the bug would still be seen as eventually we'd get a load with the same source and destination registers that would result in a register corruption failing the final state comparison but it would be preferable if these issues where spotted more reliably.

An easy way to achieve this would be to ensure the exception handler in the tests saves and restores register values and then just identify the exception handler executions in the ibex log file and remove them for the purposes of the comparison. I believe we'll be doing the save/restore anyway as the final state compare would fail otherwise.

GregAC, 2021-04-16:
Any thoughts @udinator?

GregAC, 2021-04-16:
I would note for this issue in particular we could have a simple assertion that checks we don't repeat instructions unless we've specifically jumped back to the same instruction.

udinator, 2021-04-24:
hi @GregAC - yes, I think I understand what the issue here is.
At the end of the day this is really because our testbench can't do instruction by instruction checks, which means small subtle issues like this can slip in unnoticed.

I think your proposed solution seems logical, with one caveat.
The only handlers that would be necessary to strip out of the log files would be the interrupt and debug handlers - this will need a bit of scripting work to detect and filter out the correct code sections.

I think this solution is slightly better than just using an assertion to check for this edge case, as it then provides better confidence in all other tests that involve irqs/debugs.

GregAC, 2021-04-26:
> The only handlers that would be necessary to strip out of the log files would be the interrupt and debug handlers - this will need a bit of scripting work to detect and filter out the correct code sections.

Indeed, I was thinking the testbench could output an extra log giving details of exception entry and exit (in particular including the reason for the exception) which could be used strip out handlers before comparison.

An alternative, though one that requires more work, would be using spike in lockstep as part of the simulation, which may allow it to simulate the interrupt/debug handlers too. Here we'd use spike as a library, telling it to step when Ibex steps and comparing results every retired instruction. We'd need to do some investigation to see how amenable spike would be to this in its current form (clearly it can be modified but we'd rather not have a highly modified custom spike fork).

udinator, 2021-04-27:
Yup, I think the option you suggested makes sense to me.

RE the lockstep ISS: I think this goes back to an earlier discussion we had about our preference to have some sort of cycle-accurate (or instruction-accurate) model for the core. I don't think Spike is designed around this principle, so I'd expect it to be a fairly high amount of work, and we'd end up with a highly customized fork which could potentially cause a bunch of compatibility issues down the road (as well as raising the issue of how we keep the new fork synced with the remote upstream).

GregAC, 2021-12-10:
We've now introduced a co-simulation setup using spike which provides us with instruction by instruction checking for the entirety of a test's execution.

closed 2021-12-10 by GregAC