lowRISC/ibex #1343: [dv] Consider a seperate memory access checker

GregAC, 2021-04-16:
Is it worth introducing a new checker that looks at the transactions on the data bus and confirms they all should have been executed? Such a check may help us spot problems like the one highlighted in #1339.

The question is what exactly it should check. It could record all transactions seen and we could do a post process step that each transaction matches to an executed instruction in the log, including matching up with when the instruction was executed. However this wouldn't have spotted the particular issue in #1339 in the reproduction of the bug I did as two loads were output into the log where it should have output one (see #1342).

We could drop the checking of time for the transactions and just match them up to the loads/stores reported in the spike log (i.e. just ensuring global order matches), though this will run into difficulties where we want to check tests that have exception handler execution that doesn't occur in spike (e.g. due to an interrupt).

Perhaps we are satisfied the instruction trace compare suffices here and could just add some assertions/tracking that ensures that if we see a data request on the bus this relates to a load instruction that is being reported on the trace interface.

GregAC, 2021-04-16:
Any thoughts @udinator?

udinator, 2021-04-24:
Please correct me if i'm misunderstanding, but wouldn't introducing log comparison updates like the ones discussed in #1342 mean that this checker could be unnecessary?

If we do introduce that sort of log checking enhancements, then we'd automatically be able to check all memory transactions have been executed correctly (including stores, as the memory addresses could then later be randomly accessed by load instructions).
I think then we'd only be missing checking for any load/store instructions executed by exception/irq/debug handlers, which I tried to avoid including in the riscv-dv implementations.

does that sort of make sense, or am I not quite understanding what you're describing here?

GregAC, 2021-04-26:
The current log comparison doesn't actually look at a memory trace at all, it just compares load and store instructions. I'm proposing something that snoops the actual data bus and confirms all transactions seen are correct. However I think we could do it in a lightweight way that uses the instruction tracing. E.g. during runtime we just confirm loads/store seen match exactly to the instructions reported on the trace interface then the post process compare will confirm those load/store instructions should have been executed.

Whilst the randomly generated loads and stores should eventually find a bug where a store gets malformed or missed it does make it harder to track down the bug itself (as you'll just see a mismatch at a later point) plus increases the chance you miss it altogether. The probability of seeing the bug is the probability of finding it multiplied by the probability of generating a later load that causes a mismatch that produces a failure. Better the probability of failure is just the probability of finding the bug.

Relying on the instruction comparison also won't see double load/double store cases where a transaction gets repeated. For normal memory in a single core system this is relatively harmless (though may cause performance issues). When loading/storing to non-idempotent memory (i.e. a device) this could be a big problem (two pops/pushes to a HW fifo rather than one for instance). It may cause memory consistency issues should Ibex be used in a multi-processor system also.

udinator, 2021-04-27:
Ahh, I understand what you mean now - this would be an actual snooping component for the data bus itself.

I think that's a good idea, like you mentioned we could just take advantage of the trace interface to easily check that everything matches up, and double check with the post comparison steps.

This clearly also patches up a small hole in the memory checking like #1339 showed.

GregAC, 2021-12-10:
The new co-simulation environment contains a memory access checker.

closed 2021-12-10 by GregAC