lowRISC/ibex #1352: Ibex Simple system fails to simulate with verilator

ganoam, 2021-05-03:
Ibex simple system verilator simulation fails with segmentation fault. I have tracked down the fault to the function `DpiMemUtil::RegisterMemoryArea`, and I believe the fault occurs somewhere here:
https://github.com/lowRISC/ibex/blob/8ef06de73daa7c22400f0e617088aeea06a2601c/vendor/lowrisc_ip/dv/verilator/cpp/dpi_memutil.cc#L329-L338

Unfortunately I lack the context to fix it....

I am simulating with verilator v. 4.200.
How to reproduce:
```
fusesoc --cores-root=. run --target=sim --setup --build lowrisc:ibex:ibex_simple_system # build simulator
make -C examples/sw/simple_system/hello_test/ # build elf
./build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/Vibex_simple_system --meminit=ram,examples/sw/simple_system/hello_test/hello_test.elf # Run sim
```
command to run:

rswarbrick, 2021-05-17:
Ok, I can't reproduce this and I don't think anyone else is seeing it either. Let's close this issue, but please feel free to reopen with more information if you crash into anything like this again.

mrdiogodias, 2021-05-19:
Hello,
I'm facing the same problem. I have verilator 4.028 and I just followed the commands on the readme of the simple_system directory.

Error:
`[1]    49657 segmentation fault (core dumped)  ./build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/Vibex_simple_system `

imphil, 2021-05-19:
Can you:
* Try with the latest version of Verilator (4.202)
* Run under gdb by prefixing `gdb --args` to the command you were trying to run (e.g. `gdb --args ./build/lowrisc_ibex_ibex_simple_system_0/sim-verilator/Vibex_simple_system`). Type `r` (and ENTER) after GDB starts up, and once the program crashes, type `bt` (and ENTER) to get a backtrace. Please post the backtrace here.

mrdiogodias, 2021-05-19:
It worked with the latest version of verilator (4.202).
Sorry for my laziness, I should have tried it first.

rswarbrick, 2021-05-19:
That's helpful, thanks. If you can reproduce with an older Verilator, could you give us a backtrace? It might be there's still a bug in our code that's only triggered with certain setups. It would be nice to track down what's going on.

rswarbrick, 2021-05-19:
Thanks for the bug report. I've just pushed up the PR above which will fix it in OpenTitan. Once that's merged, I'll vendor the change back into Ibex.

PR #1360 (2021-05-25): Update lowrisc_ip to lowRISC/opentitan@7117c349d
Update code from upstream repository
https://github.com/lowRISC/opentitan to revision
7117c349d5465b5152d3bb774079013924a3e9ba

* [dv/common] Improve coverage exclusion method (Cindy Chen)
* [prim] Waive unused parameter warnings for an FPGA-specific param
  (Rupert Swarbrick)
* [dpi] Fix indexing bug in ecc32_mem_area (Rupert Swarbrick)
* [dv, xcelium] Dump covergroup report (Srikrishna Iyer)
* [dvsim] Testplanner refactor & covergroup support (Srikrishna Iyer)
* [otp_ctrl] Workaround for generated prim depending on generated prim
  (Michael Sc [... cut]
closed 2021-05-25 by rswarbrick