lowRISC/ibex #1355: Is it possible to synthesis the top without the clock_gating option

ThibaultRicher, 2021-05-18:
Hello,
Actually I make some synthesis of the top module core (ibex_top) using register files based on flip-flop because for some reason in my company we don't use D-Latch. The problem is when I start the synthesis (on Genus from Cadence) it stopped because there is an error due to the inferred of a D-Latch.
So is it possible and is there no problem to do a synthesis of the ibex_top without clock_gating package etc?

Best regards Thibault

## My Environment

I using Ibex for my company who is designing ASIC ( and we don't using Latch )

**EDA tool and version:**

Hdl designer/Cadence and Mentor tools to walk on a specific design flow

**Operating system:**

CentOS

**Version of the Ibex source code:**

The last one

GregAC, 2021-05-18:
Hi @ThibaultRicher 

You can synthesize Ibex with the flip flop based register file, just include https://github.com/lowRISC/ibex/blob/master/rtl/ibex_register_file_ff.sv in the synthesis as opposed to https://github.com/lowRISC/ibex/blob/master/rtl/ibex_register_file_latch.sv

ThibaultRicher, 2021-05-18:
Hi @GregAC 

And thank you for the quick answer. I did synthesis with this register files ibex_register_file_ff.sv, but when I do the synthesis of  https://github.com/lowRISC/ibex/blob/master/rtl/ibex_top.sv I add the different package and differents module like: the https://github.com/lowRISC/ibex/blob/master/dv/uvm/core_ibex/common/prim/prim_clock_gating.sv and https://github.com/lowRISC/ibex/blob/master/dv/uvm/core_ibex/common/prim/prim_pkg.sv and this one prim_generic_clock_gating.sv and I got an error that say I used to enable latch.

GregAC, 2021-05-18:
Ah that would be the top-level clock gate which can enable/disable the clock into Ibex. Functionally it's not required so you could either edit `ibex_top.sv` to remove it (just delete the instantiation and replace with `assign clk = clk_i`) or provide your own `prim_clock_gating.sv` which just passes the clock straight through and ignores the `test_en_i` and `en_i` inputs.

rswarbrick, 2021-05-21:
@GregAC: Is this a common design requirement, do you think? In which case, maybe we either need to provide a parameter or at least document what's going on?

GregAC, 2021-05-21:
We could add something to docs explaining `prim_clock_gating` is an example implementation and implementers may want to substitute with their own primitive (almost certainly required for ASIC cases as you'll need the specific primitive from the library you're using) or switch it to a plain passthrough.

imphil, 2021-05-21:
We had something along those lines in the past: https://github.com/lowRISC/ibex/commit/6ca325fa70231944005befebeafc45ec1894b755#diff-98e5f6b20b6cc5e00c3385d909b67c849ebc32c6d59ed8d8d776e9b8f4778cfb

Ideally, users provide their own primitive library for their target technology (as we do in OT). But explaining that in Ibex is probably too much for now, so just mentioning "replace it with something that works for you" is probably a good idea.

ThibaultRicher, 2021-05-25:
Thank you again for all your answers
In first step I by pass the clock gating with the solution you gave it to me assign clk = clk_i and then I bypassed the clock gating. All doing good so.

In second step it's a simple question because I'm very curious and this is it : 
Why do you use a D-latch for clock gating and not a simply AND gate ? Like with a signal coming out from your OR beetween the two enable (test_en and ena_i) and in the second input of the AND gate you branch your clk_i then in output of this AND you got your clk_o?

Best regards,
Thibault Richer

GregAC, 2021-06-01:
Hi @ThibaultRicher apologies for the slow follow up.

The latch is needed to stop clock glitches. If you use an AND gate alone then when the enable goes high it might generate a spurious clock pulse. The latch prevents this issue, Googling 'clock gating latch' brings up some explanatory material, e.g. https://anysilicon.com/the-ultimate-guide-to-clock-gating/ and http://www.idc-online.com/technical_references/pdfs/electronic_engineering/Clock_Gating.pdf

closed 2021-12-10 by GregAC