lowRISC/ibex #1358: I tried to build the example program with this build command :make build-arty-100 program-arty in the ibex repository and I am getting a below specified error

AkhilBR98, 2021-05-20:
## My Environment

**EDA tool and version:**

**Operating system:**

**Version of the Ibex source code:**

AkhilBR98, 2021-05-20:
akhil@uhdct3-OptiPlex-3040:~/ibex$ make build-arty-100 program-arty
cd examples/sw/led && make
make[1]: Entering directory '/home/akhil/ibex/examples/sw/led'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/akhil/ibex/examples/sw/led'
fusesoc --cores-root=. run --target=synth --setup --build \
        lowrisc:ibex:top_artya7 --part xc7a100tcsg324-1
Traceback (most recent call last):
  File "/home/akhil/.local/bin/fusesoc", line 7, in <module>
    from fusesoc.main import main
  File "/home/akhil/.local/lib/python3.5/site-packages/fusesoc/main.py", line 28, in <module>
    from fusesoc.coremanager import CoreManager, DependencyError
  File "/home/akhil/.local/lib/python3.5/site-packages/fusesoc/coremanager.py", line 16, in <module>
    from fusesoc.core import Core
  File "/home/akhil/.local/lib/python3.5/site-packages/fusesoc/core.py", line 9, in <module>
    from fusesoc.capi2.core import Core as Capi2Core
  File "/home/akhil/.local/lib/python3.5/site-packages/fusesoc/capi2/core.py", line 908, in <module>
    for backend in get_edatools():
  File "/home/akhil/.local/lib/python3.5/site-packages/edalize/__init__.py", line 27, in get_edatools
    return [get_edatool(pkg) for pkg in walk_tool_packages()]
  File "/home/akhil/.local/lib/python3.5/site-packages/edalize/__init__.py", line 27, in <listcomp>
    return [get_edatool(pkg) for pkg in walk_tool_packages()]
  File "/home/akhil/.local/lib/python3.5/site-packages/edalize/__init__.py", line 18, in get_edatool
    return getattr(import_module('{}.{}'.format(__name__, name)),
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/home/akhil/.local/lib/python3.5/site-packages/edalize/apicula.py", line 3, in <module>
    from edalize.edatool import Edatool
  File "/home/akhil/.local/lib/python3.5/site-packages/edalize/edatool.py", line 11, in <module>
    from jinja2 import Environment, PackageLoader
  File "/home/akhil/.local/lib/python3.5/site-packages/jinja2/__init__.py", line 5, in <module>
    from .bccache import BytecodeCache as BytecodeCache
  File "/home/akhil/.local/lib/python3.5/site-packages/jinja2/bccache.py", line 61
    self.code: t.Optional[CodeType] = None
             ^
SyntaxError: invalid syntax
Makefile:74: recipe for target 'build-arty-100' failed
make: *** [build-arty-100] Error 1

rswarbrick, 2021-05-20:
This looks like your Python installation has something wrong: it looks like you're trying to invoke an old version of Python (python 2 maybe?) on `/home/akhil/.local/lib/python3.5/site-packages/jinja2/bccache.py`, which uses syntax that isn't supported.

imphil, 2021-05-20:
Python 3.5 isn't supported by Ibex any more, please switch to Python 3.6+.

rswarbrick, 2021-05-21:
As Philipp said, this looks like it's just a tool versioning problem. Please re-open with more information, or open a new issue if you still have problems with Python >= 3.6.

closed 2021-05-21 by rswarbrick