lowRISC/ibex #1359: I have installed the fusesoc into the lowrisc directory and getting an error which is specified in below comment

Saqlinahamad, 2021-05-20:
akhil@uhdct3-OptiPlex-3040:~/Saqlinahamad/lowRISC/fusesoc$ pip3 install -e .
Obtaining file:///home/akhil/Saqlinahamad/lowRISC/fusesoc
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/home/akhil/Saqlinahamad/lowRISC/fusesoc/setup.py", line 48, in <module>
        python_requires=">=3.5, <4",
      File "/home/akhil/.local/lib/python3.5/site-packages/setuptools/__init__.py", line 153, in setup
        return distutils.core.setup(**attrs)
      File "/usr/lib/python3.5/distutils/core.py", line 108, in setup
        _setup_distribution = dist = klass(attrs)
      File "/home/akhil/.local/lib/python3.5/site-packages/setuptools/dist.py", line 441, in __init__
        k: v for k, v in attrs.items()
      File "/usr/lib/python3.5/distutils/dist.py", line 281, in __init__
        self.finalize_options()
      File "/home/akhil/.local/lib/python3.5/site-packages/setuptools/dist.py", line 748, in finalize_options
        for ep in sorted(eps, key=by_order):
      File "/home/akhil/.local/lib/python3.5/site-packages/setuptools/dist.py", line 747, in <lambda>
        eps = map(lambda e: e.load(), pkg_resources.iter_entry_points(group))
      File "/home/akhil/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2450, in load
        return self.resolve()
      File "/home/akhil/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2456, in resolve
        module = __import__(self.module_name, fromlist=['__name__'], level=0)
      File "/home/akhil/Saqlinahamad/lowRISC/fusesoc/.eggs/setuptools_scm-6.0.1-py3.5.egg/setuptools_scm/__init__.py", line 15, in <module>
        from .version import format_version, meta
      File "/home/akhil/Saqlinahamad/lowRISC/fusesoc/.eggs/setuptools_scm-6.0.1-py3.5.egg/setuptools_scm/version.py", line 42
        trace(f"tag '{tag}' parsed to {result}")
                                              ^
    SyntaxError: invalid syntax

rswarbrick, 2021-05-20:
Looks like the same underlying cause as https://github.com/lowRISC/ibex/issues/1358, which you seem to have posted a few minutes ago with a different GitHub username. Please close one or the other of these issues so we can track the problem in just one place.

The problem looks like it's probably caused by a wonky Python installation. As a first check, run `python3 --version` and check that you see at least 3.5.

Saqlinahamad, 2021-05-20:
i'll check that

closed 2021-05-20 by Saqlinahamad